aritchie/bluetoothle #79: Adapter Status Unknown (Iphone 8, iOS 11)

mahesh139, 2017-10-11:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS 11 (IPhone 8)

## Version of Library
5.0.0

## Expected Behaviour
`CrossBleAdapter.Current.Status` must return "PoweredOn" and `CrossBleAdapter.Current.Scan `must not break with  Unknown status error.

## Actual Behavior
when i get status of the adapter using `CrossBleAdapter.Current.Status` , i get status "Unknown" even 
when bluetooth is turned on.  the problem occurs with iPhone 8 (iOS 11). i tried turning bluetooth off and on again but same problem.

aritchie, 2017-10-11:
It is likely a permission setup issue of some sort as I simply cast the value that is given from apple.  I'll need more info if I'm going to be able to help at all.

mahesh139, 2017-10-13:
thank you very much  @aritchie   for your quick reply.

I applied permissions based on your documentation (for android only.)  for android it worked perfectly fine without any problem.

Other apps (e.g. nRF connect, BlueFruite connect) were scanning perfect on the same iPhone (8, iOS 11) and there was no issue with permission.

i further investigated and found this old issue:
[](https://github.com/aritchie/bluetoothle/issues/34)

comments from you and @SuhanKoh helped me. i was  starting the scan immediately on app start.
scanning seems to be working perfect without any problem after i started using `ScanWhenAdapterReady`
`CrossBleAdapter.Current.ScanWhenAdapterReady().Subscribe(this.OnScanResult);`

thanks for this great plugin and all the help. 

Regards,
Mahesh

closed 2017-10-13 by aritchie