aritchie/bluetoothle #84: stop search in background after turn off and turn on bluetooth again

ipon85, 2017-10-18:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
6.0.1

## Version of Library
5.0.0

## Expected Behaviour
I'm using android service(Android.App.Service) for scan ble devices in background. When bluetooth is on, and service has started, it discover ble devices. Then, I hide main application to background, and turn off bluetooth. After that, I turn on bluetooth, but ble devices aren't found now. Until call main app foreground. Then search work correct as at start.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-10-18:
It sounds like a permission issue or a limitation being applied by the OS.  The library doesn't do anything special with regards to background BLE on android

ipon85, 2017-10-19:
App has permissions:
- android.permission.BLUETOOTH_ADMIN
- android.permission.BLUETOOTH
- android.permission.BLUETOOTH_PRIVILEGED
- android.permission.ACCESS_COARSE_LOCATION
- android.permission.ACCESS_FINE_LOCATION

On the first start bluetooth and moving app to background search work correct. The trouble starts when in background BT is off and then on again. On debug I see that service is working, but .ScanOrListen().Subscribe(scanResult =>{ ... } find nothing

`using System;
using System.Threading.Tasks;
using Android.App;
using Android.Bluetooth;
using Android.Content;
using Android.OS;
using Plugin.BluetoothLE;

namespace Droid
{
    [Service]
public class BleWorker : Service
{
    public override IBinder OnBind(Intent intent)
    {
        return null;
    }    
    
    public override StartCommandResult OnStartCommand(Intent intent, StartCommandFlags flags, int startId)
    {
            Plugin.BluetoothLE.IAdapter adapter = CrossBleAdapter.Current;        
        
            base.OnStartCommand(intent, flags, startId);
        
            adapter.WhenStatusChanged().Subscribe(statusC =>
            {
                try
                {
                    if (_runTask == false && statusC == AdapterStatus.PoweredOn)
                    {
                        _runTask = true;
                        RunAsyncBleSearch(adapter);
                    }
                    if (statusC == AdapterStatus.PoweredOff)
                    {
                        _runTask = false;
                    }
                }
                catch (Exception ex)
                {
                }
            });
            return StartCommandResult.Sticky;        
    }
        
    static bool _runTask = false;    

    private async void RunAsyncBleSearch(IAdapter ad)
    {
        while (_runTask)
        {
             
            try
            {
                
                if (!ad.IsScanning)
                {
                    
                    ad.ScanOrListen().Subscribe(scanResult =>
                    {
                        
                        var btd = (BluetoothDevice)scanResult.Device.NativeDevice;
                        
                        var devName = btd.Name;

                        if (!string.IsNullOrWhiteSpace(devName) )
                        {
                            //some actions
                        }
                    });
                }
                
                
            }
            catch (Exception ex)
            {

            }
            finally
            {
                await Task.Delay(4000);
            }
        }
    }

    public override void OnDestroy()
    {
        _runTask = false;
        base.OnDestroy();
    }
}

}`

aritchie, 2017-10-19:
Scans will be killed if the Bluetooth adapter is turned off.  The OS likely isn’t allowing a scan to be started in the background.  The fact that your code works in the foreground tells me that this is not an issue with the library

closed 2017-10-21 by aritchie