aritchie/bluetoothle #85: Devices not found on Android

alland67, 2017-10-19:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Api 24 on my Galaxy S7 Edge

## Version of Library
v5.0.0

## Expected Behaviour
Devices should be found. If I scan from within the device's Settings/Bluetooth page I see the other devices.

## Actual Behavior
There is no error or exception thrown. I just don't get anything back. It is being called.
Here is the snippet I am using in my core pcl for the project:

var scanner = CrossBleAdapter.Current.Scan().Subscribe((IScanResult scanResult) =>
                {
                    devices.Add(scanResult.Device.Name);
                });

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-10-19:
This library will only search out Bluetooth le devices

aritchie, 2017-10-19:
Also make sure your adapter is ready and turned on using other methods found on this library.  I am closing this because this has either been answered several times elsewhere or is documented.  

If you submit issues in the future, please include parts of your code as well.  Simply reiterating stuff I’ve written is not helpful

closed 2017-10-19 by aritchie