aritchie/bluetoothle #86: BLE peripheral with custom service UUID is discovered with message - Invalid UUID Detect - E73EA580EB2911E5AA9A0002A5D5C51B

jkmandhan, 2017-10-20:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
6.0
Works fine with iOS and UWP

## Version of Library
5.0

## Expected Behaviour
BLE peripheral should be discovered on scanning with correct ServiceGuids. It is showing up perfectly when scanning on iOS and UWP

## Actual Behavior
When trying to scan on Android platform, BLE peripheral is not being discovered correctly, The logs in Output window showing the following message - 

Invalid UUID Detect - E73EA580EB2911E5AA9A0002A5D5C51B

ServiceUuids in AdvertisementData is coming as Guid.Empty

_If applicable, please include entire exception - type, message, stacktrace_

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-10-20:
What is the proper UUID that you are expecting to be parsed here?

jkmandhan, 2017-10-20:
In the ble peripheral, Service UUID is e73ea580-eb29-11e5-aa9a-0002a5d5c51b

commit 7f2acd33 (2017-10-21): fix #86 - bad uuid size - watch for 32 char strings
closed 2017-10-21 by aritchie