aritchie/bluetoothle #87: Scan with config has only 1 result per peripheral

wmcainsh, 2017-10-23:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS

## Version of OS(s) listed above with issue
iPhone 6s - 11.0.3

## Version of Library
5.0.0

## Expected Behaviour
BleAdapter.Scan(scanConfig) - returns multiple ScanResults for each device.

## Actual Behavior
BleAdapter.Scan(scanConfig) - returns a single ScanResult for each device.

Is this how the plugin should work? I was expecting BleAdapter.Scan(scanConfig) to work the same as BleAdapter.Scan() except only return devices with the specified Service Id's.

aritchie, 2017-10-23:
Take a look at the apple docs for this one as I'm not sure.  I'm definitely not doing anything special.

aritchie, 2017-10-29:
A PR in the future would be fantastic please.

closed 2017-10-29 by aritchie