aritchie/bluetoothle #89: Uncaught exception when cancelling connection with pending characteristics read/write

goiditdev, 2017-11-17:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
7.0

## Version of Library
NuGET 5.2.0

## Expected Behaviour
There is a clean way of starting an unawaited read or write on characteristics, and when the connection fails or a CancelConnection on the IDevice is issued between the start of the read/write and the indication of completion by android througt the GattCallback no exception is thrown or it is caught internally by the plugin.

## Actual Behavior
When starting a read or write on characteristic that is not awaited, a CancelConnection on the IDevice results in an uncaught exception of a disposed semaphore. The semaphore in question is in DeviceContext.cs in Internals in public IObservable<T> LockObservable<T>() and specifically it is released on line 93 (this.semaphore.Release();).

The exception text is this:
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at md5e0c8df6e6e221d5c163c41bc679a7531.GattCallbacks.n_onCharacteristicRead(Native method)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at md5e0c8df6e6e221d5c163c41bc679a7531.GattCallbacks.onCharacteristicRead(GattCallbacks.java:38)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at android.bluetooth.BluetoothGatt$1.onCharacteristicRead(BluetoothGatt.java:334)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:117)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at android.os.Binder.execTransact(Binder.java:573)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403] Pending exception android.runtime.JavaProxyThrowable: System.ObjectDisposedException: The semaphore has been disposed.
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Threading.SemaphoreSlim.CheckDispose () [0x00013] in <183b200ee49746d48fc781625979428e>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Threading.SemaphoreSlim.Release (System.Int32 releaseCount) [0x00000] in <183b200ee49746d48fc781625979428e>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Threading.SemaphoreSlim.Release () [0x00000] in <183b200ee49746d48fc781625979428e>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at Plugin.BluetoothLE.Internals.DeviceContext+<>c__DisplayClass15_1`1[T].<LockObservable>b__1 () [0x00033] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Android\Internals\DeviceContext.cs:93 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.Disposables.AnonymousDisposable.Dispose () [0x00010] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.Disposables.SingleAssignmentDisposable.Dispose () [0x00014] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.Disposables.StableCompositeDisposable+Binary.Dispose () [0x00026] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.Disposables.SingleAssignmentDisposable.Dispose () [0x00014] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.AutoDetachObserver`1[T].Dispose (System.Boolean disposing) [0x0000a] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.ObserverBase`1[T].Dispose () [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.AutoDetachObserver`1[T].OnCompletedCore () [0x0000d] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.ObserverBase`1[T].OnCompleted () [0x0000e] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at Plugin.BluetoothLE.Extensions.Respond[T] (System.IObserver`1[T] ob, T value) [0x00007] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Abstractions\Extensions_Rx.cs:22 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at Plugin.BluetoothLE.GattCharacteristic+<>c__DisplayClass6_0.<Read>b__1 (Plugin.BluetoothLE.Internals.GattCharacteristicEventArgs args) [0x0005c] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Android\GattCharacteristic.cs:97 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.AnonymousSafeObserver`1[T].OnNext (T value) [0x0000a] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.Linq.ObservableImpl.Where`1+_[TSource].OnNext (TSource value) [0x00030] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.Observer`1[T].OnNext (T value) [0x00010] in <99f8205c51c44bb480747b577b8001ff>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at System.Reactive.Subjects.Subject`1[T].OnNext (T value) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at Plugin.BluetoothLE.Internals.GattCallbacks.OnCharacteristicRead (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.BluetoothGattCharacteristic characteristic, Android.Bluetooth.GattStatus status) [0x00000] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Android\Internals\GattCallbacks.cs:12 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at Android.Bluetooth.BluetoothGattCallback.n_OnCharacteristicRead_Landroid_bluetooth_BluetoothGatt_Landroid_bluetooth_BluetoothGattCharacteristic_I (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.IntPtr native_characteristic, System.Int32 native_status) [0x00017] in <484d528f6626418ea3cba8d2e5438e61>:0 
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at (wrapper dynamic-method) System.Object:0e5fac13-16cd-4d77-ab92-6af629b938ce (intptr,intptr,intptr,intptr,int)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at void md5e0c8df6e6e221d5c163c41bc679a7531.GattCallbacks.n_onCharacteristicRead(android.bluetooth.BluetoothGatt, android.bluetooth.BluetoothGattCharacteristic, int) (GattCallbacks.java:-2)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at void md5e0c8df6e6e221d5c163c41bc679a7531.GattCallbacks.onCharacteristicRead(android.bluetooth.BluetoothGatt, android.bluetooth.BluetoothGattCharacteristic, int) (GattCallbacks.java:38)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at void android.bluetooth.BluetoothGatt$1.onCharacteristicRead(java.lang.String, int, int, byte[]) (BluetoothGatt.java:334)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at boolean android.bluetooth.IBluetoothGattCallback$Stub.onTransact(int, android.os.Parcel, android.os.Parcel, int) (IBluetoothGattCallback.java:117)
11-17 15:28:43.705 F/art     ( 4980): art/runtime/runtime.cc:403]   at boolean android.os.Binder.execTransact(int, long, long, int) (Binder.java:573)

I will ivnestigate some fixes and report back in. 
Thanks!

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-11-18:
The issue should be fixed, however, I would recommend trapping errors coming from read/write as they can happen for many other reasons aside from this one.

closed 2017-11-18 by aritchie

goiditdev, 2017-11-18:
Thanks! I am trapping around the read/writeand yet these went uncaught. I am unsure as to why. I trapped on all exceptions, to no avail. Any suggestions? Would it help, in this case, to also trap around the CancelConnection()'s?

goiditdev, 2017-11-18:
Follow-up: This seems to, at least for me, break reads (and possibly writes) on characteristics. They now time out for me, where before they returned successfully with the read data from the characteristic. I'm investigating and hope to have more soon but my guess is that it is related to the same semaphore.

goiditdev, 2017-11-18:
It looks like 5.2.1 is no longer available on NuGET. I did previously pull it and noticed a few issues with characteristic reads and writes completing, which I believe are related to the semaphore changes. ANything I can do to help?