aritchie/bluetoothle #90: Xamarin.Forms - Can not archive for publishing a new xamarin.droid app, but xamarin.ios work.

dapoulin, 2017-11-20:
Hi,

I am facing an issue in the last part of the production of an app I have created, the publishing part.
Everything is going well for my iOs app, and I could properly generate the archive package and publish it. But, I can not create the Android Package.

It appears to be very simple to reproduce it.

Steps to reproduce

Create a blank new Xamarin.Forms project (Android & iOs)

Add to the Android application the Plugin.BluetoothLE package (5.2.0)

Switch for configuration mode to Release

Try to archive for publishing the Android app

You can not archive and got a LinkAssemblies error.

Expected behavior

I would have to be able to create the Android Package in release mode.

Actual behavior

I am getting a LinkAssemblies task failed unexpectedly exception.

Crashlog

/Users/david/Projects/BleOtherDemo/Droid/BleOtherDemo.Droid.csproj : warning XA0105: The $(TargetFrameworkVersion) for Plugin.BluetoothLE.dll (v8.0) is greater than the $(TargetFrameworkVersion) for your project (v7.1). You need to increase the $(TargetFrameworkVersion) for your project.
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: The "LinkAssemblies" task failed unexpectedly.
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: Mono.Linker.MarkException: Error processing method: 'Android.Bluetooth.BluetoothGatt Plugin.BluetoothLE.Internals.DeviceContext::CreateReflectionGatt(Java.Lang.Object)' in assembly: 'Plugin.BluetoothLE.dll' ---> Mono.Cecil.ResolutionException: Failed to resolve Java.Lang.Class[] Java.Lang.Reflect.Executable::GetParameterTypes()
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.HandleUnresolvedMethod (Mono.Cecil.MethodReference reference) [0x00006] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.MarkMethod (Mono.Cecil.MethodReference reference) [0x00049] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.MarkInstruction (Mono.Cecil.Cil.Instruction instruction) [0x00040] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.MarkMethodBody (Mono.Cecil.Cil.MethodBody body) [0x00098] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.ProcessMethod (Mono.Cecil.MethodDefinition method) [0x00162] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.ProcessQueue () [0x0001f] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: --- End of inner exception stack trace ---
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.ProcessQueue () [0x0004b] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.Process () [0x00013] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Steps.MarkStep.Process (Mono.Linker.LinkContext context) [0x0000d] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Mono.Linker.Pipeline.Process (Mono.Linker.LinkContext context) [0x00020] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at MonoDroid.Tuner.Linker.Run (Mono.Linker.Pipeline pipeline, Mono.Linker.LinkContext context) [0x00000] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at MonoDroid.Tuner.Linker.Process (MonoDroid.Tuner.LinkerOptions options, Mono.Linker.LinkContext& context) [0x00070] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Xamarin.Android.Tasks.LinkAssemblies.Execute (Java.Interop.Tools.Cecil.DirectoryAssemblyResolver res) [0x002c8] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Xamarin.Android.Tasks.LinkAssemblies.Execute () [0x00205] in <593a6fd557984367bb21e275d0fa0659>:0
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute () [0x00023] in /Users/builder/data/lanes/4992/mono-mac-sdk/external/bockbuild/builds/msbuild-15.4/src/Build/BackEnd/TaskExecutionHost/TaskExecutionHost.cs:631
/Library/Frameworks/Mono.framework/External/xbuild/Xamarin/Android/Xamarin.Android.Common.targets(1629,5): error MSB4018: at Microsoft.Build.BackEnd.TaskBuilder+d__26.MoveNext () [0x0022d] in /Users/builder/data/lanes/4992/mono-mac-sdk/external/bockbuild/builds/msbuild-15.4/src/Build/BackEnd/Components/RequestBuilder/TaskBuilder.cs:787
1 Warning(s)
1 Error(s)

Configuration

Version of the Plugin: e.g. 5.2.0

Platform: Android,

Device: e.g.

In debug mode, everything is working well (on Android and on iOS).

aritchie, 2017-11-20:
This library is compiled for android 8.  If you take a look at your error message, it shows you’re compiling for v7

closed 2017-11-20 by aritchie

dapoulin, 2017-11-20:
In your documentation, you are talking about the following platforms :

- Android 4.3+
- iOS 7+
- macOS
- tvOS
- Linux (BlueZ)
- Tizen
- Windows UWP

Is there any chances to get android 4.3+ apps working or only Android 8 is supported ?

aritchie, 2017-11-20:
Please read the Xamarin documentation.  This library does support 4.3+ 

https://developer.xamarin.com/guides/android/application_fundamentals/understanding_android_api_levels/

dapoulin, 2017-11-20:
Haha, I was wrong (feeling tired for my last command). I could properly compile by changing the target version. Thanks, have a good day.