aritchie/bluetoothle #91: [Android] Read/Write/Notification produces GATT133 or deadlocks

aritchie, 2017-11-21:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
All

## Version of Library
All

## Expected Behaviour
That the library allows async read/writes/notification requests without said issue

## Actual Behavior
As per subject

rmenezes, 2017-11-22:
@aritchie do you know what may be causing this problem? I have two characteristics, one characteristic doesn't support Write/Notify and works well, the other one supports Notify and crash after reconnect.

aritchie, 2017-11-22:
yes.  you will have to be patient for a fix, trap your error (as suggested in the previous message), or downgrade to 5.1.  In any case, this is the issue to wait on.

rmenezes, 2017-11-22:
@aritchie downgrade to 5.1 not solve my issue. Also I can't trap the errors.

rmenezes, 2017-11-22:
@aritchie I don't know how much information you need to this issue but increasing my interval from 1 second to 2 seconds (using ReadInterval) "solve" the problem.

aritchie, 2017-11-22:
@rmenezes let's not hijack this particular issue.  I'm not sure what your issue is currently because I have no error at this point.  The issue with the semaphore will only exist in 5.2 as that is where it was released.  The rest of this issue has to do with why the semaphore exists in the first place.

rmenezes, 2017-11-23:
@aritchie weird ... I downgrade to 5.0 and all the problems with semaphore is gone. Do you have any branch for the 5.1 version?

closed 2017-11-24 by aritchie

mariusbloemhof, 2017-12-12:
@aritchie, was this issue closed because it was fixed? I am still experiencing this problem when writing to characteristics, but only when writing to a characteristic in CharacteristicResult on notification received. If you want I can open a new issue and attach all the relevant information?

aritchie, 2017-12-12:
@mariusbloemhof I closed it because I can no longer reproduce this issue with any of my devices.  

For your new issue, first try leaving a 100ms pause before issuing the Write.  I'm trying to synchronize everything into the gatt, but I issue the pause "post" operation.  You are writing before I can apply that protection.

aritchie, 2017-12-12:
Actually, I lie - I am issuing the pause before.  There is a new property visible to Android only - CrossBleAdapter.AndroidOperationPause - try setting that to TimeSpan.FromMilliseconds(100);

mariusbloemhof, 2017-12-12:
Thanks, I tried that now which seems to have solved my initial write deadlock (a). but after that it remains.

**scenario (a)** _before setting `CrossBleAdapter.AndroidOperationPause` (or setting it to 0ms):_

(1) I obtain a characteristic;
(2) I can write to it without any problems repeatedly.
(3) I subscribe to notifications on this characteristic;
(4) When I get a notification, I  now want to write to the characteristic using `CharacteristicResult.Characteristic` from the notification. this fails and is deadlocked

**scenario (b)** _setting `CrossBleAdapter.AndroidOperationPause` to 100ms:_

(1) - (3) as in scenario (a)
(4) is successful, but
(5) any subsequent writes to the same `CharacteristicResult.Characteristic` as in (4) fails and is deadlocked.

setting `CrossBleAdapter.AndroidOperationPause` has definitely made (4) work in where it previously failed. I tried up to 500ms, but with the same result

mariusbloemhof, 2017-12-15:
@aritchie, do you need me to log a new issue, or can this one be reopened?

aritchie, 2017-12-23:
@mariusbloemhof no, it's ok.  I was on it right away.  There was nothing further to track since the area of issue was well known.  There was a big find as to what was causing the latest lock issue anyhow.  You are welcome to try master or dev (v6 has breaking changes) with the latest update to see how that goes.

commit dd349976 (2017-11-24): fix #91 - force sequential read/writes