aritchie/bluetoothle #94: WhenServiceDiscovered() throwing NRE on device reconnection

tplooker, 2017-11-27:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 6.0

## Version of Library
5.2.2

## Expected Behaviour
Second time connecting to device it should reconnect and rediscover the services

## Actual Behavior
My code works fine on first connection to the device I have the following method 

`public void SetDevice(ScanResult result)
 {
            if (result == null)
            {
                System.Diagnostics.Debug.WriteLine("Setting Device Result to null");
                _serviceDiscovered?.Dispose();
                _statusChanged?.Dispose();
            }
            else
            {
                System.Diagnostics.Debug.WriteLine($"Setting Device Result to scan result {result.Name}");
                DeviceObj = result.Device;

                _serviceDiscovered?.Dispose();
                _serviceDiscovered = DeviceObj.WhenServiceDiscovered()
                                            .Subscribe(service =>
                                            {
                                                if (service == null)
                                                    return;
                                                System.Diagnostics.Debug.WriteLine($"Service discovered with UUID : {service.Uuid}");
                                                setService(service);
                                            });

                _statusChanged?.Dispose();
                _statusChanged = DeviceObj.WhenStatusChanged()
                                        .Subscribe(x =>
                                        {
                                            System.Diagnostics.Debug.WriteLine("BLE Connection changing : " + x.ToString());
                                            DeviceConnectionStatus = (DeviceConnectionStatus)x;
                                        });

                LoggingStatus = false;
                WifiStatus = false;
            }
}`

which fires when a matched scan result is found to a naming convention for a device. The following method is called when I want to connect to the device.

'public bool DeviceConnect()
{
            System.Diagnostics.Debug.WriteLine("Device.DeviceConnect()");
            if (_deviceconnection != null)
            {
                System.Diagnostics.Debug.WriteLine("Existing device connection, disposing now");
                _deviceconnection.Dispose();
                _deviceconnection = null;
            }

            try
            {
                _deviceconnection = DeviceObj.Connect().ToTask();
            }
            catch (Exception e)
            {
                System.Diagnostics.Debug.WriteLine(e);
                return false;
            }
            return true;
}'

On first connection this works fine and the disconnect with this updated library now works as intendened, however when I try to reconnect i get the following NRE which is related to the WhenServiceDiscovered() method.

11-28 10:54:45.138 I/MonoDroid(26308): UNHANDLED EXCEPTION:
11-28 10:54:45.148 I/MonoDroid(26308): System.NullReferenceException: Object reference not set to an instance of an object.
11-28 10:54:45.148 I/MonoDroid(26308):   at Plugin.BluetoothLE.Device.<WhenServiceDiscovered>b__19_4 (Plugin.BluetoothLE.ConnectionStatus _) [0x00017] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Android\Device.cs:184 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.AnonymousSafeObserver`1[T].OnNext (T value) [0x0000a] in <99f8205c51c44bb480747b577b8001ff>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Linq.ObservableImpl.Where`1+_[TSource].OnNext (TSource value) [0x00030] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Linq.ObservableImpl.RefCount`1+_[TSource].OnNext (TSource value) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Subjects.FastImmediateObserver`1[T].EnsureActive (System.Int32 count) [0x0016b] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Subjects.FastImmediateObserver`1[T].EnsureActive () [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Subjects.ReplaySubject`1+ReplayBase[T].OnNext (T value) [0x0006e] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Subjects.ReplaySubject`1[T].OnNext (T value) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Linq.ObservableImpl.AsObservable`1+_[TSource].OnNext (TSource value) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Linq.ObservableImpl.DistinctUntilChanged`2+_[TSource,TKey].OnNext (TSource value) [0x00087] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Linq.ObservableImpl.Concat`1+_[TSource].OnNext (TSource value) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.AutoDetachObserver`1[T].OnNextCore (T value) [0x00002] in <99f8205c51c44bb480747b577b8001ff>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.ObserverBase`1[T].OnNext (T value) [0x0000d] in <99f8205c51c44bb480747b577b8001ff>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at Plugin.BluetoothLE.Device+<>c__DisplayClass17_0.<WhenStatusChanged>b__2 (Plugin.BluetoothLE.Internals.ConnectionStateEventArgs args) [0x0001e] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Android\Device.cs:142 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.AnonymousSafeObserver`1[T].OnNext (T value) [0x0000a] in <99f8205c51c44bb480747b577b8001ff>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Linq.ObservableImpl.Where`1+_[TSource].OnNext (TSource value) [0x00030] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Observer`1[T].OnNext (T value) [0x00010] in <99f8205c51c44bb480747b577b8001ff>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at System.Reactive.Subjects.Subject`1[T].OnNext (T value) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at Plugin.BluetoothLE.Internals.GattCallbacks.OnConnectionStateChange (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.GattStatus status, Android.Bluetooth.ProfileState newState) [0x00000] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Android\Internals\GattCallbacks.cs:57 
11-28 10:54:45.148 I/MonoDroid(26308):   at Android.Bluetooth.BluetoothGattCallback.n_OnConnectionStateChange_Landroid_bluetooth_BluetoothGatt_II (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.Int32 native_status, System.Int32 native_newState) [0x0000f] in <c8cab2efd85f442389020057942eca89>:0 
11-28 10:54:45.148 I/MonoDroid(26308):   at (wrapper dynamic-method) System.Object:929ecf73-7001-4a3d-8fe9-e02729c45f53 (intptr,intptr,intptr,int,int)

aritchie, 2017-12-04:
Fixed in v5.3

tplooker, 2017-12-04:
Thanks @aritchie  I can confirm this works for me!

closed 2017-12-04 by aritchie