aritchie/bluetoothle #97: ArgumentException: Invalid State when starting server

ChrisMash, 2017-12-02:
## Version of OS(s) listed above with issue
iOS 11.1.1 (not managed to test on Android yet)

## Version of Library
5.2.2

## Expected Behaviour
Server starts successfully

## Actual Behavior
ArgumentException: Invalid State - Unknown, thrown from server.Start(adData)

Took the example code from the GitHub Readme and haven't edited it significantly, the final code is as follows:
```
server.Start(new Plugin.BluetoothLE.Server.AdvertisementData
                    {
                        LocalName = "TestServer"
                    });
```
BLE state is PoweredOn, added the background modes (not sure I need them as likely won't be using the BLE in the background but can't hurt for now) and the permission string to the info.plist.

forestmaster22, 2018-02-05:
Is there any way to work around this issue or do I have to wait for v6.0?

aritchie, 2018-02-05:
@forestmaster22 You're welcome to contribute fixes.  I have yet to track this one down.

mphipps1, 2018-02-14:
I am having the same issue. Pretty much straight copy of the sample code. Anything that I can do to help fix?

bangonkali, 2018-07-30:
Also encountering the same issue on iPhone :(

commit 6952ff3e (2018-08-07): Fix #97 - iOS needs a second for the adapter to be ready