aritchie/bluetoothle #98: Reading or writing to a characteristic kills all further discovery

pdonovan, 2017-12-04:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

Android 6.0

## Version of Library

5.2.2, 5.3.0 (works in 5.1.0 but then I'm hit by bug #88 )

## Expected Behaviour

Reading from a characteristic during discovery should not stop all other characteristics being discovered after that.

## Actual Behavior

When connected to a particular BLE device we're supporting, if in the subscription to IGattService.GetKnownCharacteristics() you attempt to read or write to the characteristic, no other characteristics will get discovered.

The device is found by scanning for a particular service type:
`this._scan = this.BleAdapter.Scan(new ScanConfig()
                {
                    ScanType = BleScanType.LowLatency,
                    ServiceUuids = new List<Guid>() { TemperatureServiceUuid }
                })
                .Subscribe(this.OnScanResult);`

and the read or write is similar to ( see full attachment):
`
               var readValue = await characteristic.Read();

                if (readValue != null && readValue.Data != null)
                 {
                            var bytesvalue = readValue.Data;
                            byte val = bytesvalue[0];
                            System.Diagnostics.Debug.WriteLine("Auto Off mins: " + val);
                            byte autoOffSetting = 20;
                            if (val != autoOffSetting)
                            {
                                if (characteristic.CanWriteWithoutResponse())
                                {
                                    System.Diagnostics.Debug.WriteLine("Writing autoOff without response");
                                    characteristic.WriteWithoutResponse(new byte[] { autoOffSetting });
                                }
                                else if (characteristic.CanWriteWithResponse())
                                {
                                    System.Diagnostics.Debug.WriteLine("Writing autoOff with response");
                                    var writeResult = await characteristic.Write(new byte[] { autoOffSetting });
                                    System.Diagnostics.Debug.WriteLine("Writeresult " + writeResult.Data != null ? writeResult.Data[0].ToString() : "???");
                                }
                            }
                        }
`

I have noticed that each of the device's characteristics gets discovered multiple times (5 or 6 times). I'm not sure if that's expected or part of the problem.  

I've attached a log from the Android device:
[bt_log.txt](https://github.com/aritchie/bluetoothle/files/1525744/bt_log.txt)
and the full code by my BLE interaction in a test project:

[BleDeviceManager.cs.txt](https://github.com/aritchie/bluetoothle/files/1525747/BleDeviceManager.cs.txt)

aritchie, 2017-12-04:
You are likely hitting an android race condition.  The communication with Bluetooth is actually serial/sequential so if you do overlapping operations such as a read during a multi discovery process - things like this will happen.  My suggestion is request all of your known characteristics first (API take an array of them) and then perform read/writes.  I take care of making sure read/writes/notifications don’t suffer from the serial problem.

I will look into a fix in a future release, but the suggestion above will work just fine (it’s how I use it in a few projects).

As for whenservicediscovered, it was designed to repeat what has been discovered per connection & per hook to it.

pdonovan, 2017-12-04:
Thanks for the information.
So should I keep a track (dictionary?) of all the characteristics I've asked for and mark when they've been discovered? Will that be a safe way of knowing I've seen them all and can start interrogating? Or should I put some kind of additional wait time (ugh) in there as well before I start interacting with them?

pdonovan, 2017-12-05:
I'm still hot having much luck I'm afraid. Unless I fill my code with Task.Delay() calls the second Read() always gets stuck at the lock in DeviceContext.Lock<T> : [Debug][Device] Lock - at the gate.

I'm only making one call to GetKnownCharacteristics on one service, asking for 4 characteristics. When each one is found I add it to a ConcurrentDictionary. When the key count is 4 I then start reading them sequentially.

`var measurementScaleGuid = Guid.Parse(BleDeviceManager.MEASUREMENT_SCALE_CHARACTERISTIC_UUID);
var measurementTimingGuid = Guid.Parse(BleDeviceManager.MEAS_TIMING_CHARACTERISTIC_UUID);
var autoOffGuid = Guid.Parse(BleDeviceManager.AUTO_OFF_CHARACTERISTIC_UUID);
var asciiTemperatureGuid = Guid.Parse(BleDeviceManager.ASCII_TEMP_CHARACTERISTIC_UUID);

this.Device
     .GetKnownCharacteristics(BleDeviceManager.TemperatureServiceUuid,
        measurementScaleGuid,
        measurementTimingGuid,
        autoOffGuid,
        asciiTemperatureGuid
     )
     .Subscribe(async characteristic =>
     {
         System.Diagnostics.Debug.WriteLine("Found a temperature charac: " + characteristic.Uuid);
         if (characteristic.Uuid.Equals(measurementScaleGuid))
         {
             _requiredCharacteristics.AddOrUpdate(characteristic.Uuid, characteristic, (key, existing) => { return existing; });
         }
         else if (characteristic.Uuid.Equals(measurementTimingGuid))
         {
             _requiredCharacteristics.AddOrUpdate(characteristic.Uuid, characteristic, (key, existing) => { return existing; });
         }
         else if (characteristic.Uuid.Equals(autoOffGuid))
         {
             _requiredCharacteristics.AddOrUpdate(characteristic.Uuid, characteristic, (key, existing) => { return existing; });
         }
         else if (characteristic.Uuid.Equals(asciiTemperatureGuid))
         {
             _requiredCharacteristics.AddOrUpdate(characteristic.Uuid, characteristic, (key, existing) => { return existing; });
         }

         if (_requiredCharacteristics.Keys.Count == RequiredKeyCount)
         {
             await SetupCharacteristics();
         }
     });`

And in SetupCharacteristics():

`await Task.Delay(5 * 1000);

IGattCharacteristic autoOffCat = _requiredCharacteristics[Guid.Parse(BleDeviceManager.AUTO_OFF_CHARACTERISTIC_UUID);
CharacteristicResult readResult = await autoOffCat.Read();
if (readResult != null && readResult.Data != null)
{
    System.Diagnostics.Debug.WriteLine("Read auto-off (idle time before light sleep)");

    var bytesvalue = readResult.Data;
    byte val = bytesvalue[0];

    System.Diagnostics.Debug.WriteLine("Auto Off mins: " + val);

}

//await Task.Delay(3 * 1000);

IGattCharacteristic timingCat = _requiredCharacteristics[Guid.Parse(BleDeviceManager.MEAS_TIMING_CHARACTERISTIC_UUID)];
readResult = await timingCat.Read();
if (readResult != null && readResult.Data != null)
{
    System.Diagnostics.Debug.WriteLine("Read measurement timing");
    var bytesvalue = readResult.Data;

    byte val = bytesvalue[0];

    System.Diagnostics.Debug.WriteLine("Measurement timing: " + val);
}

//await Task.Delay(3 * 1000);

IGattCharacteristic scaleCat = _requiredCharacteristics[Guid.Parse(BleDeviceManager.MEASUREMENT_SCALE_CHARACTERISTIC_UUID)];
readResult = await scaleCat.Read();
if (readResult != null && readResult.Data != null)
{
    System.Diagnostics.Debug.WriteLine("Read measurement scale");
    string currScale = Encoding.UTF8.GetString(readResult.Data, 0, readResult.Data.Length);
    System.Diagnostics.Debug.WriteLine("Current scale is " + currScale);
}`

It never logs [Device] [Releasing the sync lock] unless I pause between reads.

`[0:] Found a temperature charac: 28ef0770-3640-4565-a99b-d07cf6a9103f
[0:] Found a temperature charac: 4134a840-cbfa-455a-a3a9-6b78cc64dc72
[0:] Found a temperature charac: 7d53f0b7-33aa-4939-98c6-39fcd1fa9b06
[0:] Found a temperature charac: 78544003-4394-4fc2-8cfd-be6a00aa701b
12-05 13:28:48.394 D/Mono    (23334): Assembly Ref addref BluetoothTest.Std[0xe7b500a0] -> System.Reactive.Linq[0xe7b50940]: 5
Thread finished:  #5
The thread 'Unknown' (0x5) has exited with code 0 (0x0).
[0:] [Debug][Device] Lock - at the gate
[0:] [Debug][Device] Lock - past the gate
12-05 13:28:53.459 D/BluetoothGatt(23334): readCharacteristic() - uuid: 7d53f0b7-33aa-4939-98c6-39fcd1fa9b06
12-05 13:28:53.459 D/BluetoothDevice(23334): mAddress: 00:07:80:A6:3A:7C
12-05 13:28:53.562 D/BluetoothGatt(23334): onCharacteristicRead() - Device=00:07:80:A6:3A:7C UUID=7d53f0b7-33aa-4939-98c6-39fcd1fa9b06 Status=0 srvcType=0 srvcInstId=0 charInstId=0
12-05 13:28:53.562 D/BluetoothDevice(23334): mAddress: 00:07:80:A6:3A:7C
12-05 13:28:53.562 D/BluetoothDevice(23334): mAddress: 00:07:80:A6:3A:7C
12-05 13:28:53.562 D/BluetoothDevice(23334): mAddress: 00:07:80:A6:3A:7C
12-05 13:28:53.562 D/BluetoothDevice(23334): mAddress: 00:07:80:A6:3A:7C
Thread started:  #12
[0:] Read auto-off (idle time before light sleep)
[0:] Auto Off mins: 20
[0:] [Debug][Device] Lock - at the gate
Thread finished: <Thread Pool> #8
The thread 'Unknown' (0x8) has exited with code 0 (0x0).`

If I uncomment the 3 second delays it works.

I still open to being told I'm doing something stupid - this is the first time I've used a Reactive library so I couldn't be doing something daft.

aritchie, 2017-12-05:
Please try to reduce the mass amounts of code you're including.  I've only got so much time for supporting this thing.  

As I said, the android API tends to lock up if you overwhelm it which you still are.  You need to find all of the characteristics first, then do all your reads/writes.  So watch the following: 
1) Subscribe with an async is a huge no-no in RX. It doesn't wait for the subscribe action to finish, thus it fires as fast as it gets events.  I protect the read/writes via the lock gates.  I'm not currently protecting the GetKnownCharacteristics.  
2) The second thing mistake is - never store characteristics in global scope.  Every disconnect will flush out the internals of those variables causing them to be invalid.

Here's what you should do for now:

```csharp
var yourCharacteristics = await this.Device.GetKnownCharacteristics(
   BleDeviceManager.TemperatureServiceUuid,
   measurementScaleGuid,
   measurementTimingGuid,
   autoOffGuid,
   asciiTemperatureGuid
)
.Take(4)
.ToList();

// now do your thing with the characteristics that are found
```

pdonovan, 2017-12-05:
Ok, after doing what you suggested in your comment above, I can get a list of the characteristics. But I still can't read them.
```
foreach(IGattCharacteristic c in yourCharacteristics)
{
    System.Diagnostics.Debug.WriteLine("will read: " + c.Uuid);
    c.Read().Subscribe(observer => { System.Diagnostics.Debug.WriteLine("did read: " + observer.Characteristic.Uuid); });
}
```
I have also tried:
```
foreach (IGattCharacteristic c in yourCharacteristics)
{
   System.Diagnostics.Debug.WriteLine("will read: " + c.Uuid);
   var readResult = await c.Read();
   System.Diagnostics.Debug.WriteLine("did read: " + readResult.Characteristic.Uuid);
}
```
Both methods lock at the second read - I only ever get one "did read:" message.

aritchie, 2017-12-06:
I have been unable to reproduce this problem.  I have a unit test based on this exact case.  In order to progress this, please do the following:

1) Provide the device make & model so I know if I'm fighting with a vendor
2) Put an "await Task.Delay(300);" after each read.  If it works, please reduce to 100 and see if it still works

If 2 works, I'll look into adding a pause in the locking mechanism.

pdonovan, 2017-12-06:
The device is a Cooper-Atkins Blue2, firmware 1.10. We are successfully using these devices in iOS using CoreBluetooth, and on Android using Plugin.BLE (except for rare crashes which only our clients can replicate on Samsung devices, hence me trying your library..).

After I posted yesterday I found that a delay of 500ms after each Read() worked. I just tried 100ms and it still works. I would suggest making it configurable as you have done for the other delays.

aritchie, 2017-12-08:
5.3.1 contains a operation wait parameter.  It also causes proper waits on GetKnownCharacteristic.  I now consider this case closed.

closed 2017-12-08 by aritchie

NandishGowdaNR, 2019-02-21:
Even after delay I'm getting value for Read characteristics. I'm reading fro custom service, where i have to write first and then read. Able to write successfully but read got lock and never come back. Can you help me on this as earliest.

aritchie, 2019-02-21:
@NandishGowdaNR please don't post in closed issues.  Secondly, you haven't given any details in order to know what issue you are having.  Lastly, please note that this is not a support site.  If you require support beyond the actual open source, samples, & docs - I do offer paid consultancy.  You can reach out to me via the email on my github profile if this is something you need.