aritchie/bluetoothle #100: Android - No platform plugin found. when deployed on device.

IosDeveloperHarsh, 2017-12-05:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] Android

## Version of OS(s) listed above with issue
5.3

## Version of Library
5.3

## Expected Behaviour
It should work on android when deployed.

## Actual Behavior
When we install application on the android device it gives below error.

Unhandled Exception:

System.ArgumentException: [Plugin.BluetoothLE] No platform plugin found.  Did you install the nuget package in your app project as well? occurred

## Code of Conduct
I am using latest version on the plugin that is 5.3.
I am using PCL structure in my project.
At the time of coding i am able to access the library but when building it and installing on the device its showing No platform plugin found. 
Can you Please guide me.

aritchie, 2017-12-05:
This library uses .NET standard 2, not PCL.  Please follow the setup guide on the readme.

closed 2017-12-05 by aritchie

IosDeveloperHarsh, 2017-12-06:
Can you give me the link for the readme guide Or Steps how to setup this library using .NET standard 2 because i have used nuget, installed this plugin in all of my project.

aritchie, 2017-12-06:
https://github.com/aritchie/bluetoothle/blob/master/readme.md or scroll down on the root page of this project.  Seriously though, I would consider taking a class on Xamarin University and reading up on this stuff.  I can't do .NET 101 support here, sorry.