aritchie/bluetoothle #99: Connect() doesn't connect to device and thus never emits an OnNext()

ichpuchtli, 2017-12-05:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 7.0 API 24

## Version of Library
5.3

## Expected Behaviour
Connect() connects to the device and emits at least once to signal this fact.

## Actual Behavior
Connect() never connects to the device and thus never emits (onNext's), funnily enough the WhenDeviceStatusChanged() emits when I switch the BLE Scanner app and connect to device then all the services reads etc.. work as well. I just can't get it to Connect with you Connect() method :( 

I've noticed your example in the readme doesn't subscribe to the Connect()?  Shouldn't Connect() be lazy to match the Rx paradigm and therefore require a subscriber to actually begin a connection. If so I would have expected that you would only find services/characteristics once the connect() observable had emitted an OnNext()'s or OnCompleted() to signal a connection was established. 

The code below is how I interpreted the API i.e you used Connect()'s observable to be notified when the connection was established but continue to use the device concept to look for known services etc..
connectIntents is an ```IObservable<IDevice>``` which emits when a user clicks on a device they want to retrieve info from. 

```csharp
var subscription = connectIntents
    .Select(device => device.Connect(config).Select(_ => device))
    .Switch()
    .Select(connectedDevice =>
    {
        return connectedDevice
            .GetKnownService(ServiceUuid)
            .Select(service =>
            {
                return service
                .GetKnownCharacteristics(CharacteristicUuid)
                .Select(c => c.Read())
                .Switch()
                .Select(x => Encoding.UTF8.GetString(x.Data))
                .Select(x => JsonConvert.DeserializeObject<ObjectDto>(x));
            })
            .Switch();
    })
    .Switch()
    .Subscribe(objectDto =>
    {
        render(objectDto);
    });
```
## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

ichpuchtli, 2017-12-05:
Awesome project btw. I'm a huge Rx fan and love seeing it used like this.

aritchie, 2017-12-05:
I use await for the completable Connect() observable.  It is easier for the general community to understand the completable part.  I also manage the connection to the device internally, as opposed to pushing anything out.

If you follow the examples, the connection will work just fine

ichpuchtli, 2017-12-05:
Diggin deeper apparently the device is not 'connectable' isConnectable false on the advertisement ? Anyway everything i've tried hasn't worked. Do you recommend any other xamarin bluetooth wrappear type projects? 

```csharp
 connectIntents.Subscribe(device =>
            {
                RunOnUiThread(async () =>
                {
                    if (device.AdvertisementData.IsConnectable) {
                        await device._device.Device.Connect();
                    }

                    device.WhenServiceDiscovered().Subscribe(s =>
                    {
                        System.Diagnostics.Debug.WriteLine("Service Discovered");
                    });

                    device.WhenAnyCharacteristicDiscovered().Subscribe(d =>
                    {
                        System.Diagnostics.Debug.WriteLine("Characteristic Discovered");

                    });

                });```

aritchie, 2017-12-05:
You don't need to run on UiThread.  The library does all of that internally.  Also, you don't need to check the ad data... just connect if you know the device.  You may also want to trap any connection errors.

The question for asking for other libraries is rather rude and uncalled for here.

closed 2017-12-05 by aritchie