aritchie/bluetoothle #103: After re-connecting to the previous BLE device, get "Object reference not set to an instance of an object." when enabling characteristic notification

Zenitram-Oriaj, 2017-12-13:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [x] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 7.0 (API 24)

## Version of Library
5.3.1

## Expected Behavior
First time connection, all expected behavior works correctly. I am able to enable two characteristics from the same service.

Re-connecting, I would expect the same operation to work as before.

## Actual Behavior
When reconnecting to the same device, I am able to connect just fine, and collect all available characteristics into an observable collection. 
I can then get the desired characteristic from the collection.
But when I go to enable it (`var tmp = chr.EnableNotifications().Subscribe();`), it throws an exception: "Object reference not set to an instance of an object." even though the variable `chr` has a GattCharacteristic.

```
var chr = FindCharacteristic(ATECH.ChrTorqReport);
if (chr != null) {
  Xamarin.Forms.Device.BeginInvokeOnMainThread(() => {
    try {
      var tmp = chr.EnableNotifications().Subscribe();
      if (tmp != null) {
        _torqueReportEnabled = chr.WhenNotificationReceived().Subscribe(tr => TorqReportReceived(tr.Data));

      if (_torqueReportEnabled != null) { }
     }
    }
catch (Exception ex) { SessionLogger.ExceptionHandler(ex, "SessionManager", "EnableTorqueReports");}
}); 
```

Stack-trace:
```
 at Plugin.BluetoothLE.GattCharacteristic+<>c__DisplayClass7_0+<<EnableNotifications>b__0>d.MoveNext () [0x0005c] in C:\\dev\\acr\\bluetoothle\\Plugin.BluetoothLE.Android\\GattCharacteristic.cs:128 \n--- End of stack trace from previous location where exception was thrown ---\n  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 \n  at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <9ce90c81f389405fa2d38b0e75e8871b>:0 \n  at System.Reactive.ExceptionHelpers.Throw (System.Exception exception) [0x0000a] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Stubs+<>c.<.cctor>b__2_1 (System.Exception ex) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.AnonymousSafeObserver`1[T].OnError (System.Exception error) [0x0000e] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Linq.ObservableImpl.Defer`1+_[TValue].OnError (System.Exception error) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 \n  at System.Reactive.AutoDetachObserver`1[T].OnErrorCore (System.Exception exception) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.ObserverBase`1[T].OnError (System.Exception error) [0x0001c] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.SafeObserver`1[TSource].OnError (System.Exception error) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Linq.ObservableImpl.Throw`1+_[TResult].Invoke () [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 \n  at System.Reactive.Concurrency.Scheduler.Invoke (System.Reactive.Concurrency.IScheduler scheduler, System.Action action) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Concurrency.ImmediateScheduler.Schedule[TState] (TState state, System.Func`3[T1,T2,TResult] action) [0x00014] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Concurrency.Scheduler.Schedule (System.Reactive.Concurrency.IScheduler scheduler, System.Action action) [0x0001c] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Linq.ObservableImpl.Throw`1+_[TResult].Run () [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 \n  at System.Reactive.Linq.ObservableImpl.Throw`1[TResult].Run (System.IObserver`1[T] observer, System.IDisposable cancel, System.Action`1[T] setSink) [0x00010] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 \n  at System.Reactive.Producer`1[TSource].SubscribeRaw (System.IObserver`1[T] observer, System.Boolean enableSafeguard) [0x00071] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Producer`1[TSource].Subscribe (System.IObserver`1[T] observer) [0x0000e] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Linq.QueryLanguage+<>c__DisplayClass170_0`1[TSource].<StartAsyncImpl>b__0 (System.IObserver`1[T] observer) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 \n  at System.Reactive.AnonymousObservable`1[T].SubscribeCore (System.IObserver`1[T] observer) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.ObservableBase`1[T].Subscribe (System.IObserver`1[T] observer) [0x00050] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.ObservableExtensions.SubscribeSafe[T] (System.IObservable`1[T] source, System.IObserver`1[T] observer) [0x00024] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Linq.ObservableImpl.Defer`1+_[TValue].Run () [0x0002d] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 \n  at System.Reactive.Linq.ObservableImpl.Defer`1[TValue].Run (System.IObserver`1[T] observer, System.IDisposable cancel, System.Action`1[T] setSink) [0x00010] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 \n  at System.Reactive.Producer`1[TSource].Run (System.Reactive.Concurrency.IScheduler _, System.Reactive.Producer`1+State[TSource] x) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Concurrency.ScheduledItem`2[TAbsolute,TValue].InvokeCore () [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Concurrency.ScheduledItem`1[TAbsolute].Invoke () [0x0000d] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Concurrency.CurrentThreadScheduler+Trampoline.Run (System.Reactive.Concurrency.SchedulerQueue`1[TAbsolute] queue) [0x00040] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Concurrency.CurrentThreadScheduler.Schedule[TState] (TState state, System.TimeSpan dueTime, System.Func`3[T1,T2,TResult] action) [0x00046] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Concurrency.LocalScheduler.Schedule[TState] (TState state, System.Func`3[T1,T2,TResult] action) [0x0000e] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Producer`1[TSource].SubscribeRaw (System.IObserver`1[T] observer, System.Boolean enableSafeguard) [0x0005c] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.Reactive.Producer`1[TSource].Subscribe (System.IObserver`1[T] observer) [0x0000e] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at System.ObservableExtensions.Subscribe[T] (System.IObservable`1[T] source) [0x00023] in <99f8205c51c44bb480747b577b8001ff>:0 \n  at TorqLOGGER.Services.SessionManager+<>c__DisplayClass94_0.<EnableTorqueReports>b__0 () [0x00002] in C:\\Repos\\TorqLOGGER\\TorqLOGGER\\TorqLOGGER\\Services\\SessionManager.cs:640 
```

(The SessionManager is a singleton.)

I use the same series of methods as used in the first connection so I know that logic works.
Before disconnecting, I stop all running characteristics first.

I hope this is enough information. If not, I will try to provide you with what I can.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-12-14:
Are you reusing the characteristics you put in you collection on the reconnection?

Zenitram-Oriaj, 2017-12-14:
No. I reset the observable collection and, when I make a new connection, I re-request all the characteristics from the device.

aritchie, 2017-12-14:
You'll need to send a reproducible sample.  The only way I have been able to reproduce this is by reusing the characteristics.  Does the sample app on the device you are using also produce this issue?

Zenitram-Oriaj, 2017-12-14:
I did not have issues when running the sample app, but it is not exactly the same either in that the connection, characteristics gathering, and implementing notifications of characteristics is all automated. 

I am currently implementing some of the concepts you have implemented in the sample app. I will let you know what I find.

Zenitram-Oriaj, 2017-12-18:
I resolved the issues I was experiencing with Android.

closed 2017-12-18 by Zenitram-Oriaj

aritchie, 2017-12-18:
Glad to hear it.  Can you share what your fix was so I can potentially document it so others don't make the same error?

Zenitram-Oriaj, 2017-12-21:
While I am not %100 sure what exactly was the issue, it seems by putting every Bluetooth transaction on the main UI thread seem to resolve it.