aritchie/bluetoothle #104: Master & Server

TobiasDeVries, 2017-12-18:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

all
## Version of Library

5.3.1
## Question
Is it possible, using your library, to set up the phone as master & server rather than peripheral & server?

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-12-18:
Please read the very first page on the site - the readme.  It clearly shows "GATT Server"!!

Please note that this area is for feature requests, pull requests, and defects.  You can get answers to your questions by searching the code and reading the docs.  Thank you.

closed 2017-12-18 by aritchie

TobiasDeVries, 2017-12-18:
I'm sorry if I misused this area but I wasn't able to find another way to
ask this question.

I read the docs and I'm full aware of the fact that you can configure the
phone as server. That however was not what I wanted to know.
What I need to figure out - and couldn't so far - is whether you can
configure it as server *&* master.
The phone should scan for advertisers and connect to them while still being
the server. It should not advertise itself however. Is this possible using
your library?

Sorry for bothering you and thank you for providing this library in the
first place.

Best regards,
Tobias de Vries

Robert-David-Uballe, 2018-01-14:
https://www.youtube.com/robert.uballe@att.net/