aritchie/bluetoothle #105: possible inconsistency between isScanning and WhenScanningStatusChanged on Android

goiditdev, 2017-12-19:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 7.0, but should apply to all of them

## Version of Library
5.2.0

## Expected Behaviour
CrossBleAdapter.Current.IsScanning should match the status of the Subscribe to CrossBleAdapter.Current.WhenScanningStatusChanged()

## Actual Behavior
While CrossBleAdapter.Current.WhenScanningStatusChanged() return the scanning status based on where within public override IObservable<IScanResult> Scan(ScanConfig config) it is (calls to this.scanStatusChanged.OnNext(true); when starting and then to this.scanStatusChanged.OnNext(false); upon return), the IsScanning on android maps to IsScanning => this.manager.Adapter.IsDiscovering;, which is I understand it correctly actually is the status of a Bluetooth Device Discovery scan, not a BLE scan. My apologies if I got this wrong. In my Subscribe I can see the subscribed value showing true, but IsScanning reporting false, and subsequent calls to IsScanning always report false even with a scan underway and BLE devices being discovered: 
IDisposable subscription = CrossBleAdapter.Current.WhenScanningStatusChanged().Subscribe(scanning =>
            {
                if (scanning == true || adapter?.IsScanning == true) statereached.Set();
            });

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-12-19:
No matter what the fix is - I would check both.  On android, there is a chance that isscanning changes AFTER I fire the event.  This isn't something I would consider broken.... it is just an android thing

goiditdev, 2017-12-19:
For me, IsScanning never changes to true, as I do not do any BT device discovery scans. But I also admit that I am a bit confused with Android's wording for the description of isDiscovering (from the BluetoothAdapter). I just don't know that that state properly reflects BLE scanning or not. I hope I'm not off base here...

borbmizzet, 2018-01-11:
This seems to occur with Device.WhenStatusChanged() as well. The ConnectionStatus param passed to the observer was "Disconnected" but Device.Status was still "Connected".

Observed with Plugins.BluetoothLE 5.3.1 and Android 7.0.

aritchie, 2018-01-11:
@borbmizzet You should never check both... the event usually fires BEFORE the state changes.  This is an android thing.  There isn't really a need to check Status if you are monitoring it anyhow.

Robert-David-Uballe, 2018-01-14:
"© 2018-NA. Robert David Uballe. Forevermore. All things in part(s) or all of Robert David Uballe , robert.uballe@att net , trent.1980@yahoo.com , trent.uballe1980@iCloud.com , cute79764@yahoo.com , odessamassages@me.com , robert.uballe@gmail.com , 4323084982@mymetropcs.com , 4322880402@mms.att.net , 4322880402@txt.att.net , 4322880403@mms.att.net , trent.1980@att.net trentuballe@outlook.com and or its likeness is hereby bylaws , bypartsin man made or godly in GoodFaith I Robert David Uballe must keep safe all things private personal medical or information or facts Forevermore and all things listed or not listed must help balance that order and grant all things alotted myself gifts donations awards and granted compensation amounts."

aritchie, 2018-01-17:
This is fixed as part of upcoming v6

closed 2018-01-17 by aritchie