aritchie/bluetoothle #106: iOS App Crash on reconnect to device.

Zenitram-Oriaj, 2017-12-21:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS 10.2.1

## Version of Library
5.3.1

## Expected Behaviour
Connect and Reconnect to same device to work each time
First time connection (when app runs) works as expected.

## Actual Behavior
On my app, when reconnecting to the same previous device, the app crashes.

When trying using the provided sample app, essentially the same behavior. (some times on the first re-connect, but always the second reconnect)

App crashes as shown in the attached image.

[image]

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-12-21:
Please include the full stacktrace.  That NRE doesn't help me at all.

Also, please answer my question in #103.  I spent time looking into a potential issue for you only to have it closed without any explanation

aritchie, 2017-12-23:
You need to provide more info than this.  I have been trying to reproduce this without any luck.

aritchie, 2017-12-28:
No feedback and insufficient info.  Tests and demo app all work for me.

closed 2017-12-28 by aritchie