aritchie/bluetoothle #107: Deadlock in DeviceContext.lock

FrozenKiwi, 2017-12-21:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
4.4, 5, 6 (I think)

## Version of Library
5.3 + master branch

## Expected Behaviour
Lock should _not_ lock it's own thread (ie, it should be re-entrant safe for a single thread).

## Actual Behavior
A single thread can enter lock 2x, causing it to lock waiting for itself

A condensed callstack is as below.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-12-21:
Please include a sample or unit test.  I need to see why others are having this problem

FrozenKiwi, 2017-12-21:
I'll see if I can.  It's a pretty involved project.  I don't know if I could reproduce this in an isolated case, as well as it depends on some custom hardware (metawearC).

I'll continue digging to see if I can tease apart the root cause

FrozenKiwi, 2017-12-22:
Ok, so the cause is actually pretty straight-forward in my case.  It seems the return from the Context.lock function is not getting executed, thus never releasing the AutoResetEvent

The code calling the fn is:

        public async Task<byte[]> ReadCharacteristicAsync(Tuple<Guid, Guid> gattChar)
        {
            var ch = await GetGattCharacteristic(gattChar);
            if (ch != null)
            {
                var res = await ch.Read();
                return res.Data;
            }
            return new byte[0];
        }

If I call ReadCharacteristicAsync twice in a row, it deadlocks.

I'm not awfully familiar with reactive programming (in fact, this library is the entirety of my experience).  Any idea why the "await" does not trigger the execution of the returned lamda from "lock"?

aritchie, 2017-12-22:
I don't believe the problem is actually a deadlock.  I believe the problem is that android is tossing an error out from the backend.  I'm adding a bunch of error checks in v6 to bring other failures out to the surface

FrozenKiwi, 2017-12-22:
seems likely, but the value we read from the characteristic is consistent, so the read succeeds.  I also don't see anything in the device log to indicate an error.

This is quite an old phone (4.4.2, I think).  Not sure if that affects anything

aritchie, 2017-12-22:
It is easy to overwhelm the BLE stack on android.  The lock is an attempt to prevent that.  You can play with CrossBleAdapter.AndroidOperationPause = TimeSpan.FromMilliseconds(250);  

I default it to 100ms at the moment.  You can also try leaving a gap after each read

await Task.Delay(250);

FrozenKiwi, 2017-12-22:
Ok, so after a bit more digging: does this make sense?

We call Ch.read() =>
...
in Android.GattCharacteristic.Read, we first lock then run the following lines:

... (create an observable for the following)
                        this.Value = args.Characteristic.GetValue();

                        var result = new CharacteristicResult(this, CharacteristicEvent.Read, this.Value);
                        ob.Respond(result); // This line is executed
                        this.ReadSubject.OnNext(result); // This line is not
...

The read succeeds, we call ob.Respond, which then continues execution after my await:

        public async Task<byte[]> ReadCharacteristicAsync(Tuple<Guid, Guid> gattChar)
        {
            var ch = await GetGattCharacteristic(gattChar);
            if (ch != null)
            {
                var res = await ch.Read(); // Respond returns execution to this line
                return res.Data; // and we continue on
            }
            return new byte[0];
        }

_without ever calling this.ReadSubject.OnNext(result)_

Of course, once the IP is back in my code I call read a second time, and we grind to a halt.

I tried modifying my read as:

...
            if (ch != null)
            {
                var res = ch.Read().ToTask();
                await Task.Delay(250);
                return (await res).Data;
            }
...

but no change in behaviour

FrozenKiwi, 2017-12-22:
Ok, so I modified your lock function as follows to get past the lock

                pastGate = true;
                Log.Debug("Device", "Lock - past the gate");

                sub = inner.Subscribe(
                    ob.OnNext,
                    ob.OnError,
                    () =>
                    {
                        Log.Debug("Device", "Releasing sync lock");
                        this.reset.Set();
                        pastGate = false;
                        ob.OnCompleted();
                    }
                );

This works (kinda) as expected:

Read is performed, onNext & onComplete is called, lock is released.

Some time later, the returned lambda is triggered, calling this.reset.set() a second time.  I don't know why the lambda takes so long to execute, but this change seems in the spirit of the idea at least?

aritchie, 2017-12-22:
The strange part is that when ob.OnCompleted is called, it will run the cleanup function (the return action in the lock) which releases the lock which is likely the second release you're seeing

aritchie, 2017-12-22:
Well if that's the trick to fixing this, I'll patch it up and ship tonight.

FrozenKiwi, 2017-12-22:
>>The strange part is that when ob.OnCompleted is called, it will run the cleanup function (the return action in the lock)

The really strange part is that it doesn't seem to be doing that. It seems that it waits until this.ReadSubject.OnNext(result); is called.  

I can't test now, but my primary device seemed to work as you described (Android 7.1, Sony Z5).  Unfortunately, it's screen fell out so I can't verify this.  My gut instinct is that it could be some sort of performance difference, what I am working with now is _much_ slower.

Do the changes as given here raise any red flags for you?

aritchie, 2017-12-22:
Thanks for the help

FrozenKiwi, 2017-12-22:
yw.  I actually have a few other changes in a fork (they seem to be device-specific fixes), if you are doing a release soon I'll send through a PR?

FrozenKiwi, 2017-12-22:
PR here:  https://github.com/aritchie/bluetoothle/pull/108#pullrequestreview-85223900

I'm not sure how to assign you as reviewer, I'd guess it's automatic but notifying you here just in case.

aritchie, 2017-12-22:
I'm working on v6 with a bunch of changes.  Trying to keep patches to a minimum right now.  I'll see about throwing in the UWP stuff as well.  You're brave to be using it though.  I can't even get client side disconnection to work right now.  I'm guessing your host device boots GATT connections off pretty quickly making this usable for you?

FrozenKiwi, 2017-12-22:
I haven't paid much attention to disconnecting - my project is an always-connected system.  It seems to be working fine (and made iteration much faster than android... ugh it's painful).

aritchie, 2017-12-23:
Fixed - will be released with v5.3.2 or v6 - depends on timing in the next few days

closed 2017-12-23 by aritchie

aritchie, 2017-12-28:
v6-beta is live on nuget.  It does have minor signature changes.