aritchie/bluetoothle #109: RegisterAndNotify limit 20 bytes on Android

Uniphix, 2017-12-22:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
7.1.1

## Version of Library
5.2.2

## Expected Behaviour
RegisterAndNotify contains full data buffer of more than 20 bytes.

Basically when i step through the RegisterAndNotify, the data is 20 bytes, when on iOS it is limited to 158 bytes.  My data that I have incoming is slightly more than 20 bytes but under the 158 byte limit.  I am just trying to figure out how I can make it either send more bytes or make it so that I can read the next set of bytes within that same notification?

## Actual Behavior
RegisterAndNotify limits to 20 bytes rather than constructing buffer of more than 20 bytes.  I'd like to know how I can structure to read the next 20 bytes if needed.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2017-12-22:
This is a general limit of BLE for the most part.  You can use the adapter.RequestMtu which can give you up to 512 bytes on a limited set of devices.  You'll need to read up on the devices/OS versions that will allow you to do this.  I'm afraid that question goes beyond this library though.

Best of luck.

closed 2017-12-22 by aritchie

Uniphix, 2017-12-22:
I guess basically what i'll need to do is not use notify unless to let the device know that there it needs to do a read or something.  This is what I feared.

Also I cannot upgrade to 5.3.1 was there a reason for removing ScanListen?

aritchie, 2017-12-22:
Because it was painful to maintain so it won't back.

Uniphix, 2017-12-22:
Okay thanks that makes sense, so is there a way to read the next set of bytes on notify or will I need to do some sort of notify where it reaches back and read the set of data I need to read up to a certain amount?

Something like:

Incoming Notify -> Total Bytes = 1294 bytes, do a read based on mtu and then keep reading until you have all the buffer?  Or is there something like

notify's data = 20 bytes.  Read the next 20 bytes until you have reached all the buffer of data?  Probably not going to work that way with the current library setup?

Uniphix, 2017-12-22:
Nevermind I see what your saying about RequestMtu.  Thanks!

Uniphix, 2017-12-22:
So I attempted to change the Mtu and it worked however it still only brought back 20 bytes of data.  

https://socialdrip.s3.amazonaws.com/ble.png

aritchie, 2017-12-22:
As I said, you'll need to read up on the device & OS support!  It isn't a guarantee that you can go beyond 20 bytes.  The sending device may only be sending 20 bytes at a time anyhow.  iOS sometimes does internal buffering to make you believe it at a larger MTU.

Uniphix, 2017-12-22:
Gotcha, just what I feared.  Thanks