aritchie/bluetoothle #113: Example code doesn't work. No "GetAwaiter" definition.

ThiagoAguirre, 2018-01-09:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
5.0.1 (LG G2) and 6.0.1 (Samsung Galaxy S4)

## Version of Library
5.3.1

## Expected Behaviour
The provided examples should work
Should be possible to use await to read characteristics

## Actual Behavior
There's no definition to "GetAwaiter".
Manifest is correct.

```csharp
var device = CrossBleAdapter.Current.GetKnownDevice(new Guid("00000000-0000-0000-0000-000B57295595"));

//await device.Connect(); //<-This doesn't work(as provided in example)
device.Connect().Subscribe(); // <- This way works
```

**Characteristic discovery and reading**
```csharp
device.GetKnownService(new Guid("56f09012-06fc-43b2-b367-8cdb7e5c6caf")).Subscribe(service =>
{
	service.WhenCharacteristicDiscovered().Subscribe(characteristic =>
	{
		if (characteristic.CanRead())
		{
			//The code below returns the error that there's no "GetAwaiter" definition
			//await characteristic.Read();

			//The line below execute only once, the second time everything stops(the whole code)
			characteristic.Read().Subscribe(readData =>
			{
				//The code never reaches this part
			});
		}
	}
}
```
**I Tried to replace the above code for**
```csharp
characteristic.WhenRead().Subscribe(read =>
{
	//The code never reaches this part	
});

characteristic.Read();
```

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-01-09:
It works just fine - You need to bring in the namespace from reactive extensions:

using System.Reactive.Threading.Tasks;

closed 2018-01-10 by aritchie

ThiagoAguirre, 2018-01-12:
@aritchie Even after using System.Reactive.Threading.Tasks I can't use await as in the example.
Any suggestions?

Regards,

aritchie, 2018-01-12:
Please read the reactive extensions docs or review the sample code in the future!

using System.Reactive.Linq;
using System.Reactive.Threading.Tasks;

System.Reactive.Linq will give you the ability to simply await observable;  The System.Reactive.Threading.Tasks will give you an extension method to .ToTask();  which is better in any case as it allows you to pass a cancellation token.