aritchie/bluetoothle #116: iOS does not properly retrieve 2nd characterisitc when IDevice.GetKnownCharacterisitic is called twice.

borbmizzet, 2018-01-15:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [*] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
10.3.3

## Version of Library
6.0.0 beta1

## Expected Behaviour
Assert(guid1 != guid2)
IGattCharacteristic var1 = await IDevice.GetKnownCharacteristics(serviceGuid, guid1);
IGattCharacteristic var2 = await IDevice.GetKnownCharacteristics(serviceGuid, guid2);

assert(var1 == CharacteristicWithGuid1)
assert(var2 == CharacterisiticWithGuid2)

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
In Android:
var1 == CharacteristicwithGuid1
var2 == CharacterisiticwithGuid2

In iOS:
var1 == CharacteristicwithGuid1
var2 == CharacterisiticwithGuid1 <------Bug here.  Should be CharacterisiticwithGuid2

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

closed 2018-01-17 by aritchie

akhare22sandeep, 2018-03-05:
Please let me know which version have fixed this issue ? Right now i am using 5.3.2 and my application breaks if i upgrade to 6.0.0-beta3

aritchie, 2018-03-05:
5.3.2 should be fine.  6.0 has had breaking changes that you need to adapt to

aritchie, 2018-04-04:
@erdmenchen "Me too" in closed issues doesn't add anything tangible that I can be of any assistance on.  Please file issues with full test cases following the issue template that give me something to work with.  As far as my tests go, this works fine now.

v6 is in beta and changing a lot right now.  I only take PR's on betas, but its OSS so you can do whatever you'd like.  There will never be an upgrade guide unless it comes in the form of a PR.  

As for the last question, I feel that this should be obvious.  I do my best to keep consistency between versions.

commit 3206cceb (2018-01-17): fix #116 - check requested list, not just discovery list