aritchie/bluetoothle #118: Android not discovering services after connection

borbmizzet, 2018-01-17:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
7.0.0

## Version of Library
6.0.0-beta2

## Expected Behaviour
Device.WhenAnyCharacteristicDiscovered() observer gets called.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
Device.WhenAnyCharacteristicDiscovered() observer is never called.

I believe this is due to your Android/Internals/DeviceContext.cs:Close() method not following the solution given at [https://stackoverflow.com/questions/24025486/android-bluetooth-le-not-discovering-services-after-connection](https://stackoverflow.com/questions/24025486/android-bluetooth-le-not-discovering-services-after-connection)  I'd submit a PR, but I'm not sure how you'd prefer to wait for the state change. 

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-01-17:
I haven't been calling Disconnect for several releases due to the gatt race conditions that model produces.  I also have not observed the missing characteristic discovery or observed it in months.  You'll need to provide a bit sample code (discovery all the way through to the discovery process) as well as the device that is having the issue

borbmizzet, 2018-01-19:
Device: Samsung Galaxy S6 Edge+ SM-G928V

I should say, more specifically the procedure to get this to occur is:
```
IDisposable dis1 = Adapter.Scan(ScanConfig(ServiceGuid="xxxxxxx")).Subscribe(result1 => deviceobj1= result1.Device);
Delay(5 seconds);
dis1.Dispose();
await deviceobj1.Connect().Timeout(5); (success)
service1 = await deviceobj1.GetKnownService("xxxxxxx").Timeout(15); (success)
characteristics = await service1 .GetKnownCharacteristics("yyyy","zzzz").ToDictionary.Timeout(15); (success)
Send a command to the perepheral to reboot BLE device into some mode with a completely different GAP Profile and GATT Service. (success)
deviceobj1.Disconnect();
IDisposable dis2 = Adapter.Scan(ScanConfig(ServiceGuid="uuuuuuuu")).When(result=> result.Device.Uuid == deviceobj1.Uuid).Subscribe(result2 => deviceobj2 = result2.Device)
Delay(5 seconds)
dis2.Dispose()
await deviceobj2.Connect() (success);
service2 = await deviceobj2.GetKnownService("uuuuuuu").Timeout(20) (fail - TimeoutException)

```
 --Alternatively, no service or characteristics discovered even after 20 seconds when subscribing to their 'WhenDiscovered' observable before calling deviceobj2.Connect() .

This works on iOS reliably, but on Android it seldom succeeds.  That's not to say it never does, though

Another thing I've noticed in this scenario on Android is that result2.Device.Status often has an initial value of ConnectionStatus.Connected.  This can be observed if you store that property's value to a temporary variable in the first line of the observer function and set a breakpoint right after.  However, it seems that by the time the debugger catches up, result2.Device.Status has already changed to Disconnected.

Is that enough information?
On a side note, great work on the library.

borbmizzet, 2018-01-22:
Thanks for the clarification.  I was confused in thinking that connecting to a new GATT service required a new device object.  I was able to eliminate the second scan and use the same device object to connect to the second service after a delay.

aritchie, 2018-01-23:
You don't need any delay.  I manage all of this internally.  You can even configure the delay in your android project using CrossBleAdapter.Android....

closed 2018-01-23 by aritchie

borbmizzet, 2018-02-06:
Thanks, I didn't know about that.  That will be helpful.