aritchie/bluetoothle #119: Feature Request: change behavior of Adapter when it detects a scan already in progress

goiditdev, 2018-01-19:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
ios 10+, post-Lollipop android, but really any version

## Version of Library
based off of 5.2.0, but applies to any

## Expected Behaviour
FEATURE REQUEST:
For one of our applications we needed to achieve a specific plugin behavior: When the plugin started a scan for BLE devices and the adapter determined that there already was a scan ongoing it throws an exception: throw new ArgumentException("There is already an active scan"); This happens in the Scan function for each Adapter.
Android : 
public override IObservable<IScanResult> Scan(ScanConfig config)
        {
            if (this.IsScanning)
                throw new ArgumentException("There is already an active scan");
...

iOS: 
public override IObservable<IScanResult> Scan(ScanConfig config)
        {
            config = config ?? new ScanConfig();

            if (this.Status != AdapterStatus.PoweredOn)
                throw new ArgumentException("Your adapter status is " + this.Status);

            if (this.IsScanning)
                throw new ArgumentException("There is already an existing scan");

We modified this behavior to automatically stop the detected scan and then continue to start a new scan. It would be great if this could at least be an optional behavior (or the standard behavior).
What we are doing is this:

Android:
public override IObservable<IScanResult> Scan(ScanConfig config)
        {
            if (this.IsScanning)
            {
                this.context.StopScan();
                this.scanStatusChanged.OnNext(false);
             }

iOS: 
public override IObservable<IScanResult> Scan(ScanConfig config)
        {
            config = config ?? new ScanConfig();
 
            if (this.Status != AdapterStatus.PoweredOn)
                throw new ArgumentException("Your adapter status is " + this.Status);
 
            if (this.IsScanning)
            {
                this.context.Manager.StopScan();
                this.ToggleScanStatus(false);
            }

## Actual Behavior
Please see the description above.
Thank you very much!

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-01-19:
I’m sorry.  This isn’t something I want to do.  I actually used to have a ScanListen but have removed it in v6 due to the fact that the scan filters could change.

goiditdev, 2018-01-19:
We understand. No problem. 
I just wanted to clarify the reasoning from our side. The problem we encountered that this solves for us is the unlikely scenario that a user force-closes the app, or the OS does, in which case an ongoing scan gets abandoned and doesn't shut down. Particularly on iOS, upon next app start, when we first want to start the scan, the recovered context allows us to detect that there is an ongoing scan this.IsScanning returns true), even though we definitely did not yet start it for this app cycle. But because we also do not hold an IObservable from subscribing to the Scan we cannot shut it down from the outside. So there is no way to recover from this other than to ask the user to cycle the Bluetooth adapter status in the OS manually. It is for that purpose that we added this custom code.
If adding this is not an option, would it be possible to either expose a functin to force-stop an ongoing scan or to expose the adapter's context? This way, we could catch the exception in the app's code, and then force-stop it from the outside to recover the adapter state so that we can successfully start a scan again? Thank you for considering this!

aritchie, 2018-01-19:
On iOS, when the app is forced closed, I have never observed the scan remaining "live" on the next launch.  Are you using any backgrounding functionality?

aritchie, 2018-01-19:
From Android, the BLE adapter functions are static/shared.  I'm not really sure why it wouldn't see the sustained scan.  Are you launching the service externally to the app?

For iOS, are you using any of the built in plugin iOS backgrounding stuff on the adapter (BleAdapterConfiguration)?

You should open with the explanation next time.  I'll implement/fix things that make sense.  Your code originally suggested that you simply weren't cleaning something up somewhere.  The use-case you've presented makes much more sense and highlights something I do need to look into.

goiditdev, 2018-01-19:
My apologies. Yes, I should have and realized that I missed that based on the reasoning for closing the issue originally. I don't have a clue why we don't see that issue on Android (although I'm not complaining about that ;).
On iOS, yes, we do initialize it using: 
CrossBleAdapter.Init(BleAdapterConfiguration.DefaultBackgroudingConfig);

When we staet scanning we differentiate between background mode (when the service/longrunning task takes over) and when the app is foregrounded:

ScanConfig cfg = new ScanConfig();
cfg.ScanType = backgroundScan ? BleScanType.Background : BleScanType.LowLatency;
cfg.ServiceUuids = new List<Guid>
                {
                    SERVICEIDS.UUID_eddystone
                };
scanner = CrossBleAdapter.Current.Scan(cfg).Subscribe(scanResult =>
                        {
                            scanresult_processQueue.Enqueue(scanResult);                            
                        });

aritchie, 2018-01-25:
v6.0-beta3 adds Adapter.StopScan() as well as Adapter.ScanExtra(config, RESTART_SCAN) extensions

closed 2018-01-25 by aritchie

biapar, 2018-03-01:
@aritchie Where is reported this

> For iOS, are you using any of the built in plugin iOS backgrounding stuff on the adapter (BleAdapterConfiguration)?

aritchie, 2018-03-01:
@biapar I have no idea what you are asking me here?

biapar, 2018-03-02:
I read comments and you wrote about BleAdapterConfiguration. So I try to find some info about BleAdapterConfiguration.

goiditdev, 2018-03-02:
@biapar This only applies to iOS, where the plugin provides additional configuration options for accessing the BLE adapter. That BLEAdapterConfiguration class passed to the Init function can be viewed here: https://github.com/aritchie/bluetoothle/blob/787fd343f35e25bba0c004110c9f0655f76f6e55/Plugin.BluetoothLE.Apple.Shared/BleAdapterConfiguration.cs
It's easy to find with the github repo search tool at the top of the page.
Hope this helps!