aritchie/bluetoothle #120: Problem receiving multiple packets from notification

barmouta, 2018-01-22:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS 10.2.1 (iPhone 5S), iOS 10.3.3 (iPhone 6 Plus)

## Version of Library
5.3.1

## Expected Behaviour
I subscribe to characteristic as per doco:
```
if (characteristic.CanNotify())
            {
                _scanDataWatch = characteristic
                    .WhenNotificationReceived()
                    .ObserveOn(RxApp.MainThreadScheduler)
                    .Subscribe(GetScanDataNotificationHandler);
                characteristic.EnableNotifications(true);
            }
```
and this is working just fine. However, the particular characteristic is supposed to return multiple packets: the 1st packet is 5 bytes: 1 byte for packet number (i.e. 0) and 4 bytes for the size of data; 2nd packet is 20 bytes: 1 byte for packet number (i.e. 1) and 19 bytes of actual data; 3rd packet is 20 bytes: 1 byte for packet number (2) and the next 19 bytes of actual data; and so on until all data is returned.

## Actual Behavior
The notification is only invoked once, and returns the 1st packet only with 5 bytes, the last 4 bytes indicating that there are 3822 bytes of actual data, which is never received. Can you please tell me what I'm doing wrong? The notification is implemented as such:
```
void GetScanDataNotificationHandler(CharacteristicResult notification)
        {
            try
            {
                Console.WriteLine($"Received scan data notification; size: {notification.Data.Length}");
                if (_scanDataBuffer == null)
                {
                    _scanDataByteCount = BitConverter.ToInt32(notification.Data, 1);//note that byte 0 is the packet count reference, which we need to bypass
                    _scanDataBuffer = new List<byte>(_scanDataByteCount);
                    Console.WriteLine($"Scan data size: {_scanDataByteCount}");
                }
                else
                {
                    Console.WriteLine($"Received actual scan data; size: {notification.Data.Length}");
                    //throw away first byte, which is packet number
                    for (int index = 1; index < notification.Data.Length; index++)
                    {
                        _scanDataBuffer.Add(notification.Data[index]);
                    }

                    if (_scanDataBuffer.Count == _scanDataByteCount)
                    {
                        //TODO: deserialise buffer
                        //stop notifications
                        _scanDataWatch?.Dispose();
                        _scanDataWatch = null;
                    }
                    else
                    {
                        //Update progress in UI
                    }
                }
            }
            catch (Exception ex)
            {
                var msg = ex.Message;
            }
        }
```

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

barmouta, 2018-01-22:
Forgot to mention that the app is Xamarin.Forms and the above code is in the .NET Standard 2.0 project, if it helps.

aritchie, 2018-01-22:
are you subscribing to other characteristics as well?  If so, can you share the code around "if (characteristic.CanNotify())  " -- basically, I need to see everything from connect to your end goal

barmouta, 2018-01-23:
What is also interesting is that if in my Xamarin.iOS project I setup native listeners I also get one notification. So is this something I need to configure in the connection to the scanner device?

```
public void EnableNativeNotification(IGattCharacteristic characteristic)
        {
            if (characteristic is GattCharacteristic)
            {
                var nativeChar = ((GattCharacteristic)characteristic).NativeCharacteristic;
                var peripheral = nativeChar.Service.Peripheral;

                peripheral.SetNotifyValue(true, nativeChar);
                
                peripheral.UpdatedCharacterteristicValue += PeripheralUpdatedCharacterteristicValue;
            }
        }

void PeripheralUpdatedCharacterteristicValue(object sender, CoreBluetooth.CBCharacteristicEventArgs e)
        {
            if (e.Error != null)
            {
                throw new System.Exception(e.Error.ToString());
            }

            var uuid = e.Characteristic.UUID.ToString().ToUpper();
            var bytesValue = e.Characteristic.Value.ToArray();
        }
```

aritchie, 2018-01-23:
No offense, I don't have the time in life to sort through 300+ lines of code.  You also already have your answer anyhow.  This is something to do with the host device behavior obviously which I can't help with anyhow.

Here's a few suggestions
1) Quit doing this .ObserveOn(RxApp.MainThreadScheduler) unless you need to broadcast to the UI
2) FindServiceCharacteristic is not how you want to do things.  You can "await" the results of GetKnownCharacterstics.  It will return a list of found characteristics.
3) Your async voids are going to overwhelm the device if that is in a loop.
4) Why are you using IAdapterScanner?  I haven't documented it because you only need for UWP (which I don't officially support) and potentially macOS.  Just use IAdapter (CrossBleAdapter.Current).
5) My guess is that your data stream isn't flowing likely because it uses a TX/RX flow control.  Meaning for every packet you receive, you likely have to write a receive on an TX characteristic to let the host know to continue.

Anyhow, I'm closing this issue as I consider it external to the library.  Best of luck.

closed 2018-01-23 by aritchie