aritchie/bluetoothle #122: NullReferenceException when doing a SendWithoutResponse on Android

borbmizzet, 2018-01-22:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
6.0.1, 7.0.0, Galaxy S5, Galaxy S6 Edge+

## Version of Library
6.0.0-beta2

## Expected Behaviour
Not throwing a NullReferenceException at some point after executing the finally block in
try{
characteristic.WriteWithoutResponse();
}
catch(Exception ex)
{
}
finally{
}

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
Crashing at some point after executing the finally block above.

01-22 16:03:47.299 I/MonoDroid(11238): UNHANDLED EXCEPTION:
01-22 16:03:47.309 I/MonoDroid(11238): System.NullReferenceException: Object reference not set to an instance of an object.
01-22 16:03:47.309 I/MonoDroid(11238):   at Plugin.BluetoothLE.GattCharacteristic+<>c__DisplayClass19_0.<RawWriteNoResponse>b__0 () [0x0005b] in <1cce99afbe4940b9b4291fc9dc804d97>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at Plugin.BluetoothLE.Internals.DeviceContext+<>c__DisplayClass13_1.<Marshall>b__1 (System.Object _) [0x00000] in <1cce99afbe4940b9b4291fc9dc804d97>:0 
01-22 16:03:47.309 I/MonoDroid(11238): --- End of stack trace from previous location where exception was thrown ---
01-22 16:03:47.309 I/MonoDroid(11238):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <9ce90c81f389405fa2d38b0e75e8871b>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at System.Reactive.ExceptionHelpers.Throw (System.Exception exception) [0x0000a] in <99f8205c51c44bb480747b577b8001ff>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at System.Reactive.Stubs+<>c.<.cctor>b__2_1 (System.Exception ex) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at System.Reactive.AnonymousObserver`1[T].OnErrorCore (System.Exception error) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at System.Reactive.ObserverBase`1[T].OnError (System.Exception error) [0x0001c] in <99f8205c51c44bb480747b577b8001ff>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at System.Reactive.AutoDetachObserver`1[T].OnErrorCore (System.Exception exception) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at System.Reactive.ObserverBase`1[T].OnError (System.Exception error) [0x0001c] in <99f8205c51c44bb480747b577b8001ff>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at Plugin.BluetoothLE.Internals.DeviceContext+<>c__DisplayClass13_1.<Marshall>b__1 (System.Object _) [0x00023] in <1cce99afbe4940b9b4291fc9dc804d97>:0 
01-22 16:03:47.309 I/MonoDroid(11238):   at Android.App.SyncContext+<>c__DisplayClass2_0.<Post>b__0 () [0x00000] in <8211f1f12f9e45838d631abf63e51c79>:0 
01-22 16:03:47.319 I/MonoDroid(11238):   at Java.Lang.Thread+RunnableImplementor.Run () [0x00008] in <8211f1f12f9e45838d631abf63e51c79>:0 
01-22 16:03:47.319 I/MonoDroid(11238):   at Java.Lang.IRunnableInvoker.n_Run (System.IntPtr jnienv, System.IntPtr native__this) [0x00008] in <8211f1f12f9e45838d631abf63e51c79>:0 
01-22 16:03:47.319 I/MonoDroid(11238):   at (wrapper dynamic-method) System.Object:b68401c3-e1c4-4b4f-8741-54e3d6c1b273 (intptr,intptr)

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-01-22:
Please reply to other issues you've opened before opening new ones!

borbmizzet, 2018-01-25:
added a bit of extra information about my use case.

aritchie, 2018-01-31:
This is fixed as part of beta 3 which I'll be releasing in a day or two

closed 2018-01-31 by aritchie

borbmizzet, 2018-02-06:
Awesome, thanks.

commit cdc07e38 (2018-01-27): New writewithoutresponse observable to handle apple ready event and android queue mechanics.  Should fix #122 - though was never tested since this method was being updated anyhow