aritchie/bluetoothle #123: Unable to read HR and Battery Level

nicolgit, 2018-01-22:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 7

## Version of Library
5.3

## Expected Behaviour
I have to read both HR value and battery level from a specific device. I am not sure my code is correct, so I need your advice: with the following snippet I am able to read HR values but I am not receiving any battery data (row "_log($"BATTERY VALUE...") in never reached.

`
IGattCharacteristic _hrCharacteristic;
IGattCharacteristic _batteryCharacteristic;
CrossBleAdapter.Current.Scan().Subscribe<IScanResult>(scanResult =>
	{
		if (scanResult.Device.Name == _deviceName)
		{
			_device = scanResult.Device;
			var svc = _device.Connect();
			svc.Subscribe(observer =>
			{
				scanResult.Device.WhenAnyCharacteristicDiscovered().Subscribe(anyObserver =>
				{
					if (anyObserver.Uuid.Equals(CHARACTERISTIC_HEARTRATE))
					{
						_hrCharacteristic = anyObserver;

						anyObserver
							.WhenReadOrNotify(TimeSpan.FromSeconds(3))
							.Subscribe(result =>
							{
								// removed code required to read HR data HERE to simplify sample
								_log($"HEARTRATE DATA AVAILABLE!");
								
								if (_batteryCharacteristic != null)
								{
									_batteryCharacteristic.Read();
									if (_batteryCharacteristic.Value != null && _batteryCharacteristic.Value.Count()>0)
									{
										_log($"BATTERY VALUE: {_batteryCharacteristic.Value[0]}");
									}
								}
							}, cts.Token);

						anyObserver.EnableNotifications();
					}
					if (anyObserver.Uuid.Equals(CHARACTERISTIC_BATTERYLEVEL))
					{
						_batteryCharacteristic = anyObserver;
						_log($"BATTERY Level FOUND!");
					}
				}, cts.Token);
			});
		}
	},
	cts.Token);
`

aritchie, 2018-01-22:
You need to either await or subscribe to you Read() commands.  Same with EnableNotifications.  If that still isn't working for you, try 5.3.2 which will be out later today or 6.0-beta2

nicolgit, 2018-01-23:
I have tested also subscribing the read event, with no success.

I am testing 5.3.2 as soon as you release it, and come back here with a feedback!

thank you!
Nicola

nicolgit, 2018-01-23:
Hi Aritchie, I have upgraded to 5.3.2 and I have fixed my code as below (hope to have understood your hint)

`CrossBleAdapter.Current.Scan().Subscribe<IScanResult>(scanResult =>
{
	if (scanResult.Device.Name == _deviceName)
	{
		_device = scanResult.Device;
		var svc = _device.Connect();

		svc.Subscribe(observer =>
		{
			scanResult.Device.WhenAnyCharacteristicDiscovered().Subscribe(anyObserver =>
			{
				if (anyObserver.Uuid.Equals(CHARACTERISTIC_HEARTRATE))
				{
					_hrCharacteristic = anyObserver;

					anyObserver
						.WhenReadOrNotify(TimeSpan.FromSeconds(3))
						.Subscribe(result =>
						{
							_log($"HEARTRATE VALUE2: {bpm.ToString()}");

							if (_batteryCharacteristic != null)
							{
								_batteryCharacteristic.Read();
								
							}
						}, cts.Token);

					anyObserver.EnableNotifications();
				}
				if (anyObserver.Uuid.Equals(CHARACTERISTIC_BATTERYLEVEL))
				{
					_batteryCharacteristic = anyObserver;
					_batteryCharacteristic.WhenRead().Subscribe(Result =>
					{
						if (Result != null && Result.Data != null && Result.Data.Count() > 0)
						{
							_log($"BATTERY VALUE: {Result.Data[0].ToString()}");
						}
					});
					_log($"BATTERY Level FOUND!");
				}
			}, cts.Token);
		});
	}
},
cts.Token);
`

but the line log($"BATTERY VALUE: {Result.Data[0].ToString()}"); is never reached.
(no problem for the HR value)

Am I missing something?

aritchie, 2018-01-25:
You're definitely not doing this in an ideal way.  Several suggestions:

1) you really shouldn't store characteristics in variables as stated in the FAQ.  They do not survive across reconnections
2) I wouldn't scan filter by device name - it isn't ideal.  You should use service UUIDs instead
3) WhenRead() is a secondary event, it is not a resulting event.  Basically, don't use it!  It is gone in v6 anyhow.  Use Read() and await or subscribe to its result.
4) WhenReadOrNotify is a looping read call (if notify attribute is not present).  You can use this if you want to poll the characteristic at a regular interval. 

Try something like the following:
```csharp
CrossBleAdapter.Current.Scan().Where(x => x.Device.Name?.Equals("whatever")).Take(1).Select(x => x.Device).Subscribe(dev=>
{
     // you don't need to be connected to hook this and it will fire with each reconnection
     dev.WhenAnyCharacteristicDiscovered().Subscribe(ch => 
    {
        // if one of your UUIDs - CHARACTERISTIC_HEARTRATE
        ch.RegisterAndNotify().Subscribe(async data => {
            // do your read logic
            var result = await otherCharacteristic?.Read();
            // do something
        });
    });
    await dev.Connect(); // don't manage the connection yourself after this.  only call disconnect when you are done with the device
});
```

In any case, I have not reproduced this.  I'll leave it open for another day or two if you can provide any additional feedback if my suggestions don't work for you.

nicolgit, 2018-01-25:
thank you arichie!
trying to follow up tonight,N

aritchie, 2018-01-31:
no response - closing

closed 2018-01-31 by aritchie

nicolgit, 2018-02-08:
Sorry for the late answer... it **WORKS**! 

Thank you!