aritchie/bluetoothle #124: Null Reference Exception when Subscribing a second time on Android

cunnza, 2018-01-23:
## platforms

 - iOS (iPhone 6s)
 - Android (Samsung Galaxy S7)

## Version of OS(s) listed above with issue
- iOS 11.2.1
-Android 7.0 

## Version of Library
Tested on 5.3.1 and 6.0.0betaV2

## Expected Behaviour
1. Connect to Device and Subscribe to its notify Characteristic.
3. Disconnect Device ( disabling notifications before)
4. Connect again and Subscribe again

## Actual Behavior
At step nr. 4 Connecting works as expected but when i try to Subscribe again to the same device,
following Exception is coming up:

````
Unhandled Exception:

System.NullReferenceException: Object reference not set to an instance of an object.

Thread finished: <Thread Pool> #9
The thread 0x9 has exited with code 0 (0x0).
01-23 13:39:47.730 D/BluetoothGatt( 5480): readRssi() - device: 00:13:43:0D:02:XX
01-23 13:39:47.741 D/Mono    ( 5480): DllImport attempting to load: '/system/lib64/liblog.so'.
01-23 13:39:47.745 D/Mono    ( 5480): DllImport loaded library '/system/lib64/liblog.so'.
01-23 13:39:47.745 D/Mono    ( 5480): DllImport searching in: '/system/lib64/liblog.so' ('/system/lib64/liblog.so').
01-23 13:39:47.745 D/Mono    ( 5480): Searching for '__android_log_print'.
01-23 13:39:47.745 D/Mono    ( 5480): Probing '__android_log_print'.
01-23 13:39:47.746 D/Mono    ( 5480): Found as '__android_log_print'.
01-23 13:39:47.756 D/AbsListView( 5480): viewType is heaer or footer
01-23 13:39:47.759 I/MonoDroid( 5480): UNHANDLED EXCEPTION:
01-23 13:39:47.799 I/MonoDroid( 5480): System.NullReferenceException: Object reference not set to an instance of an object.
01-23 13:39:47.799 I/MonoDroid( 5480):   at Plugin.BluetoothLE.GattDescriptor.NativeEquals (Plugin.BluetoothLE.Internals.GattDescriptorEventArgs args) [0x000a1] in <1cce99afbe4940b9b4291fc9dc804d97>:0 
01-23 13:39:47.799 I/MonoDroid( 5480):   at System.Reactive.Linq.ObservableImpl.Where`1+_[TSource].OnNext (TSource value) [0x00002] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
01-23 13:39:47.799 I/MonoDroid( 5480): --- End of stack trace from previous location where exception was thrown ---
01-23 13:39:47.800 I/MonoDroid( 5480):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <9ce90c81f389405fa2d38b0e75e8871b>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at System.Reactive.ExceptionHelpers.Throw (System.Exception exception) [0x0000a] in <99f8205c51c44bb480747b577b8001ff>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at System.Reactive.Stubs+<>c.<.cctor>b__2_1 (System.Exception ex) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at System.Reactive.AnonymousSafeObserver`1[T].OnError (System.Exception error) [0x0000e] in <99f8205c51c44bb480747b577b8001ff>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at System.Reactive.Linq.ObservableImpl.Where`1+_[TSource].OnNext (TSource value) [0x00017] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at System.Reactive.Observer`1[T].OnNext (T value) [0x00010] in <99f8205c51c44bb480747b577b8001ff>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at System.Reactive.Subjects.Subject`1[T].OnNext (T value) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at Plugin.BluetoothLE.Internals.GattCallbacks.OnDescriptorWrite (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.BluetoothGattDescriptor descriptor, Android.Bluetooth.GattStatus status) [0x0000e] in <1cce99afbe4940b9b4291fc9dc804d97>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at Android.Bluetooth.BluetoothGattCallback.n_OnDescriptorWrite_Landroid_bluetooth_BluetoothGatt_Landroid_bluetooth_BluetoothGattDescriptor_I (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.IntPtr native_descriptor, System.Int32 native_status) [0x00017] in <8211f1f12f9e45838d631abf63e51c79>:0 
01-23 13:39:47.800 I/MonoDroid( 5480):   at (wrapper dynamic-method) System.Object:bb17c72d-5cc1-409c-9435-cd24359a0b80 (intptr,intptr,intptr,intptr,int)
01-23 13:39:47.827 W/art     ( 5480): JNI RegisterNativeMethods: attempt to register 0 native methods for android.runtime.JavaProxyThrowable
01-23 13:39:47.831 D/Mono    ( 5480): DllImport searching in: '__Internal' ('(null)').
01-23 13:39:47.831 D/Mono    ( 5480): Searching for 'java_interop_jnienv_throw'.
01-23 13:39:47.831 D/Mono    ( 5480): Probing 'java_interop_jnienv_throw'.
01-23 13:39:47.831 D/Mono    ( 5480): Found as 'java_interop_jnienv_throw'.
An unhandled exception occured.

Thread finished: <Thread Pool> #8
The thread 0x8 has exited with code 0 (0x0).
Thread finished: <Thread Pool> #7
The thread 0x7 has exited with code 0 (0x0).
01-23 13:40:01.277 D/BluetoothGatt( 5480): readRssi() - device: 00:13:43:0D:02:XX
01-23 13:40:01.278 E/mono    ( 5480): 
01-23 13:40:01.278 E/mono    ( 5480): Unhandled Exception:
01-23 13:40:01.278 E/mono    ( 5480): System.NullReferenceException: Object reference not set to an instance of an object.
01-23 13:40:01.278 E/mono    ( 5480):   at Plugin.BluetoothLE.GattDescriptor.NativeEquals (Plugin.BluetoothLE.Internals.GattDescriptorEventArgs args) [0x000a1] in <1cce99afbe4940b9b4291fc9dc804d97>:0 
01-23 13:40:01.278 E/mono    ( 5480):   at System.Reactive.Linq.ObservableImpl.Where`1+_[TSource].OnNext (TSource value) [0x00002] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
01-23 13:40:01.278 E/mono    ( 5480): --- End of stack trace from previous location where exception was thrown ---
[... 45 lines cut ...]
                                if (data.Data.Length > 7 && data.Data[7] > 128)
                                    return;
                                foreach (var item in data.Data)
                                {
                                    ReceivedData.Add(item);
                                }
                            });
                        }
                    });
                }
                else
                {
                    throw new Exception("FATAL ERROR: ----> Notify Characteristic Not Found !!!!!!!!!!!!!");
                }
            }
            catch (Exception)
            {

                throw;
            }
        }
````
I'm really desperate right now and looking forward to any help. 
Thanks in advance! 
cunnza

cunnza, 2018-01-23:
it always crashes at this specific line:

`await notifyChar.EnableNotifications();`

i printed out following properties:
````
                    var properties = notifyChar.Properties;   // CharacterisitcProperties.Notify
                    var description = notifyChar.Description; // "" <- empty string
                    var value = notifyChar.Value;             // Null
````
Could it maybe has to do with the empty Description ??

I also get the same Exception when using  `notifyChar.RegisterAndNotify()`

aritchie, 2018-01-25:
I have not reproduced this.  I only ever see this error when people reuse characteristic objects after a reconnection.  The method you have sent isn't ideal in my opinion and I can't tell what may be going on around that method.  Below accomplishes pretty much what you need with less risk and a bit less code

```csharp
device.WhenStatusChanged().Where(x => x == ConnectionStatus.Connected).Subscribe(_ => {
    device.GetKnownCharacteristic(YOUR_SERVICE_UUID, CHARACTERISTIC_NOTIFY_ID)
       .Timeout(..)
       .Select(x => x.RegisterAndNotify())
       .Switch()
       .Subscribe(result => 
      {
          // your logic here
      });
});
await device.Connect(); // reconnections are enabled here, don't try to scan/connect again yourself
```

aritchie, 2018-01-27:
No response - closing

closed 2018-01-27 by aritchie

cunnza, 2018-01-30:
Hello Allan,

i don’t know if you get this because the thread is already closed but in case you get it:
First, thanks for the help and suggestions and sorry i didn't response, i had important business the last four days ! I tried your optimized method, but the strange thing is,
I can’t get the write Characteristic anymore. I already tried getting the write Characteristic in 
the same style like above, but It won’t work. He seems to be stuck at getting the write Characteristic.

Any Help is appreciated! 
Thanks, cunnza

aritchie, 2018-01-31:
Use 5.3.2 or 6-betaX.  there was a deadlocking issue in v5.3.0/1.

cunnza, 2018-01-31:
Yes it works now, thank you man ! but i'm still getting Gatt 115 error when disconnecting and connecting again. For now i work around this by not disconnecting and saving the connected Devices...

aritchie, 2018-01-31:
Don't manage the connection.  I do it for you under the hood which is likely what is causing you the grief.  Only call disconnect when you are done with the device.  Simply hang on to the device instance.

cunnza, 2018-02-01:
Ok, i will go on this way, many thanks for the help !