aritchie/bluetoothle #128: Problem with IGattCharacteristic.Read

Newer91, 2018-01-31:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [*] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
6.0.1
## Version of Library
6.0.0-beta2

## Expected Behaviour
Get IGattCharacteristic.Data

## Actual Behavior
 Unhandled Exception:
 System.ArgumentNullException: Value cannot be null.
 Parameter name: value
   at System.BitConverter.ToString (System.Byte[] value) [0x00003] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at App4.BLE+<<Read>b__26_0>d.MoveNext () [0x00034] in C:\Users\balisowski\source\repos\App4\App4\App4\BLE.cs:378 
 --- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_1 (System.Object state) [0x00000] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x00007] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <657aa8fea4454dc898a9e5f379c58734>:0 
 [ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentNullException: Value cannot be null.
 Parameter name: value
   at System.BitConverter.ToString (System.Byte[] value) [0x00003] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at App4.BLE+<<Read>b__26_0>d.MoveNext () [0x00034] in C:\Users\balisowski\source\repos\App4\App4\App4\BLE.cs:378 
 --- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__6_1 (System.Object state) [0x00000] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.QueueUserWorkItemCallback.WaitCallback_Context (System.Object state) [0x00007] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading.ThreadPoolWorkQueue.Dispatch () [0x00074] in <657aa8fea4454dc898a9e5f379c58734>:0 
   at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <657aa8fea4454dc898a9e5f379c58734>:0 

## Code of Conduct
Connect Code

  new Thread(async () =>
            {
                Thread.CurrentThread.IsBackground = true;
                Thread.CurrentThread.Priority = ThreadPriority.Highest;
                await Task.Run(() => getDevice());        
            }).Start();

        void getDevice()
        {
            if (CrossBleAdapter.Current.Status == AdapterStatus.PoweredOn)
            {
                if (this.IsScanning)
                    this.scan?.Dispose();
                else
                {
                    IsScanning = true;

                    this.scan = adapter
                        .Scan()
                        .ObserveOn(RxApp.MainThreadScheduler)
                        .Subscribe(this.OnScanResult);
                }
            }
        }

        void OnScanResult(IScanResult result)
        {
            var tmp1 = "B8BC";
            var tmp2 = "B7B9";

            if (result.Device.NativeDevice.ToString().Replace(":", "").Contains(tmp2))
            {
                if (IsScanning)
                {
                    this.scan?.Dispose();
                    IsScanning = false;
                }

                device = result.Device;
                if (device.Status == ConnectionStatus.Disconnected)
                {
                    using (var cancelSrc = new CancellationTokenSource())
                    {
                        device.Connect().ToTask(cancelSrc.Token).ContinueWith(task => GetCharacters());
                    }
                }
            }
        }

       public void GetCharacters()
        {
            cleanup.Add(device.WhenServiceDiscovered().Subscribe(
                service =>
                {
                    var characters = service.WhenCharacteristicDiscovered().Subscribe(character =>
                    {
                        if (character.Uuid == this.character)
                            gattCharacteristic = character;
                        if (character.Uuid == configCharacter)
                        {
                            new Thread(() =>
                            {
                                Thread.CurrentThread.IsBackground = true;
                                foreach (var parameter in parameters)
                                {
                                    character.Write(parameter).Subscribe(sub => result.Add(sub));
                                    Thread.Sleep(150);
                                }
                            }).Start();
                        }
                    });
                }));
        }

Read code

public void Read()
        {
            if (gattCharacteristic == null)
                return;

            new Thread(async () =>
            {
                Thread.CurrentThread.IsBackground = true;
                var tmp = gattCharacteristic.Value;
                tmpp2 = BitConverter.ToString(tmp);
                Debug.WriteLine(tmpp2);
                var read = await gattCharacteristic.Read().ToTask();
                tmpp2 = BitConverter.ToString(read?.Data);
                Debug.WriteLine(tmpp2);
            }).Start();
        }

What should i do to get data from device?

aritchie, 2018-01-31:
The characteristic is not returning any data.  The response can still be successful without data being returned.  The exception shows exactly where your code is crashing.

Unhandled Exception:
System.ArgumentNullException: Value cannot be null.
Parameter name: value
at System.BitConverter.ToString (System.Byte[] value) [0x00003] in <657aa8fea4454dc898a9e5f379c58734>:0
at App4.BLE+<b__26_0>d.MoveNext () [0x00034] in C:\Users\balisowski\source\repos\App4\App4\App4\BLE.cs:378

Newer91, 2018-01-31:
Ok but it should return some data, because i sanding in with other connection.
(I have connect my computer to other port using COM and i put a data to device)

aritchie, 2018-01-31:
The read result has a success flag you should also check.  Beyond this, there isn't much I can do to help here as the library is passing all of my current tests for read.  You can also switch to 5.3.2 as I usually don't offer any support for beta's...

Newer91, 2018-01-31:
Now when i switch to 5.3.2 i got error:
    System.MissingMethodException: Method 'Plugin.BluetoothLE.CrossBleAdapter.get_Current' not found

aritchie, 2018-01-31:
Clear your bin/obj folders - xamarin 101 always when weird issue popup

Newer91, 2018-01-31:
Ok i switch to 5.3.2 i run a change method:

        public void Read()
        {
            if (gattCharacteristic == null)
                return;

            var tmpdsad = gattCharacteristic.Properties;

            new Thread(async () =>
            {
                Thread.CurrentThread.IsBackground = true;
                
                var read = await gattCharacteristic.Read().ToTask();
                if (read.Characteristic.Value == null || !read.Characteristic.CanRead())
                    Debug.WriteLine("Feilure");
                tmpp2 = BitConverter.ToString(read?.Data);
                Debug.WriteLine(tmpp2);
            }).Start();
        }
I got error:
System.ArgumentException: Failed to read characteristic - ReadNotPermitted

But in 
var tmpdsad = gattCharacteristic.Properties;
i have 
Plugin.BluetoothLE.CharacteristicProperties.Indicate | Plugin.BluetoothLE.CharacteristicProperties.Notify | Plugin.BluetoothLE.CharacteristicProperties.Read | Plugin.BluetoothLE.CharacteristicProperties.Write | Plugin.BluetoothLE.CharacteristicProperties.WriteNoResponse

aritchie, 2018-01-31:
Ok listen - no offense - this is not BLE debugging 101.  If you look at the code, you'll see this is what is coming back from android ( https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE.Android/GattCharacteristic.cs#L92 ).  You'll notice that I'm not doing a thing with the result provided by the OS.  You'll need to do a bit of digging here yourself.  It is likely some incompatibility from your host device.

Newer91, 2018-01-31:
Ok thanks for help:)
Have a nice day

closed 2018-02-01 by aritchie