aritchie/bluetoothle #129: Problems connecting to the device.

urmatgit, 2018-02-01:
Hi! can't connect to the device, Run the example and then choose your device then click "Connect" and wait a very long time. what was the problem please help. And other programs and BLE scanners connected at once.

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [x] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

## Version of Library
5.3.1 and 6.0

## Expected Behaviour

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-02-01:
I use the gatt autoconnect under the hood which means it will connect to the device when android sees it.  This is usually the preferred behaviour.  You can change this behaviour in the Connect(GattConnectionConfig) to AutoConnect to false to improve this, but you need to ensure the device is nearby

closed 2018-02-01 by aritchie

urmatgit, 2018-02-01:
Thanks for the reply, but I tried all the options GattConnectionConfig AutoConnect and the priority of two devices to be Radom 50cm-10cm does not work

aritchie, 2018-02-01:
There is nothing more I can do beyond this.  I only know that setting the autoConnect true (default) makes the connection slow.  Setting it to false will make it fast but the device has to be there or it will error.  There is nothing I can do beyond this that I know of

urmatgit, 2018-02-02:
If I AutoConnect to false do then there is here this error
[image]

aritchie, 2018-02-02:
You have to catch these errors.  This is an issue with Android that I have tried everything humanly possible to deal with and the fact is that you can only do so much aside from simply trying again