aritchie/bluetoothle #130: Multiple connections crash if disconnect other device

ianewby, 2018-02-02:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Nouget and Oreo

## Version of Library
5.2.0 (Need .net standard 1.2 unfortunately)

## Expected Behaviour
We have an application which lists bluetooth LE devices which we manufacture. We can scan fine, connect to the devices and get data from them. If we have multiple devices in range, these work fine until we disconnect from a device.

ie if we have two connected devices, the system works fine.
If we disconnect from one of them, we can reconnect to the one which is disconnected and everything is fine.
If we try to connect to the other device which is already connected the system crashes. The device is showing as connected, but we get an null reference error in the log when we try to write to a Characteristic.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
02-02 12:01:45.577 I/MonoDroid( 6723): UNHANDLED EXCEPTION:
02-02 12:01:45.641 I/MonoDroid( 6723): System.NullReferenceException: Object reference not set to an instance of an object.
02-02 12:01:45.641 I/MonoDroid( 6723):   at Plugin.BluetoothLE.GattCharacteristic.NativeEquals (Plugin.BluetoothLE.Internals.GattCharacteristicEventArgs args) [0x0005d] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Android\GattCharacteristic.cs:241 
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Reactive.Linq.ObservableImpl.Where`1+_[TSource].OnNext (TSource value) [0x00002] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
02-02 12:01:45.641 I/MonoDroid( 6723): --- End of stack trace from previous location where exception was thrown ---
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <657aa8fea4454dc898a9e5f379c58734>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow (System.Exception exception) [0x00006] in <9ce90c81f389405fa2d38b0e75e8871b>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Reactive.ExceptionHelpers.Throw (System.Exception exception) [0x0000a] in <99f8205c51c44bb480747b577b8001ff>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Reactive.Stubs+<>c.<.cctor>b__2_1 (System.Exception ex) [0x00000] in <99f8205c51c44bb480747b577b8001ff>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Reactive.AnonymousSafeObserver`1[T].OnError (System.Exception error) [0x0000e] in <99f8205c51c44bb480747b577b8001ff>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Reactive.Linq.ObservableImpl.Where`1+_[TSource].OnNext (TSource value) [0x00017] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Reactive.Observer`1[T].OnNext (T value) [0x00010] in <99f8205c51c44bb480747b577b8001ff>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at System.Reactive.Subjects.Subject`1[T].OnNext (T value) [0x00000] in <e9c1ccec51844dbd92b833a0b4bc960e>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at Plugin.BluetoothLE.Internals.GattCallbacks.OnCharacteristicChanged (Android.Bluetooth.BluetoothGatt gatt, Android.Bluetooth.BluetoothGattCharacteristic characteristic) [0x00000] in C:\dev\acr\bluetoothle\Plugin.BluetoothLE.Android\Internals\GattCallbacks.cs:22 
02-02 12:01:45.641 I/MonoDroid( 6723):   at Android.Bluetooth.BluetoothGattCallback.n_OnCharacteristicChanged_Landroid_bluetooth_BluetoothGatt_Landroid_bluetooth_BluetoothGattCharacteristic_ (System.IntPtr jnienv, System.IntPtr native__this, System.IntPtr native_gatt, System.IntPtr native_characteristic) [0x00017] in <28e323a707a2414f8b493f6d4bb27c8d>:0 
02-02 12:01:45.641 I/MonoDroid( 6723):   at (wrapper dynamic-method) System.Object:8c53bda9-7459-4ceb-8aaa-3ed793500c24 (intptr,intptr,intptr,intptr)

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-02-02:
This usually only happens when you pin a characteristic across connections.  Can you share the code?

ianewby, 2018-02-02:
The scan method:

ianewby, 2018-02-02:
I'll see if I can simplify it on Monday so I can post it all. Thanks for your help.

ianewby, 2018-02-05:
Hi,
I've tried to simplify the code down to the basics - its in the attached zip file.

[SealSentry.UI.zip](https://github.com/aritchie/bluetoothle/files/1694585/SealSentry.UI.zip)

aritchie, 2018-02-10:
You'll likely need to update.  There was a few defects around known characteristic discovery somewhere in that version timeframe.

ianewby, 2018-02-12:
Hi,
I have updated to the latest version through nuget (5.3.2) and the issue is still there. If I dont disconnect, it works fine. As soon as I disconnect the first sensor, the second detected sensor crashes.

aritchie, 2018-02-12:
I've added some safeties to the latest beta3.  Please give it a try, if not you'll need to do some digging from your side to see what the issue could be.  I don't have this use-case or a way to reproduce it at the moment.

ianewby, 2018-02-12:
Hi,
The beta3 seems to have solved the issue. Thanks for your help.

closed 2018-02-12 by aritchie