aritchie/bluetoothle #131: Issue when Disconnecting / reconnecting to device.

MDLSKF, 2018-02-08:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 7.1.1

## Version of Library
6.00-beta2

## Expected Behaviour
Hi and thanks a lot for this library!
Our sensor has 23 characteristics. After disconnecting and reconnecting the device (IDevice) gives 46 characteristics in WhenAnyCharacteristicDiscovered().Subscribe(async characteristic =>..
next time it is 69 characteristics and so on.

My sensor class has a List<IGattCharacteristic>. When connecting and discovering all characteristics I populate this list:

device.WhenAnyCharacteristicDiscovered().Subscribe(async characteristic =>
                {
                    characsFound++;
                    Debug.WriteLine(string.Empty, "Characteristics found: " + characsFound);
                    if (characteristics.Count < totalCharacteristicsCount && !characteristics.Contains(characteristic))
                    {
                        characteristics.Add(characteristic);
                    }

                    if (characteristics.Count == totalCharacteristicsCount)
                    {
                        await SetupAllCharacteristics();
                    }

 After all characteristics is discovered I start enabling notifications and read / write etc to the characteristics:

foreach (IGattCharacteristic caracteristic in characteristics)
                {
                    await Task.Delay(50);
                    switch (caracteristic.Uuid.ToString().ToUpper())
                    {
                        case DEVICE_NAME_CHAR:
                            caracteristic.Read().Subscribe(result =>
                            {
                                Name = System.Text.Encoding.UTF8.GetString(result.Data);
                            });
                            break;

                        case MANUFACTURER_NAME_CHAR:
                            caracteristic.Read().Subscribe(result =>
                            {
                                Manufacturer = System.Text.Encoding.UTF8.GetString(result.Data);
                            });
                            break;
                       case BATTERY_LEVEL_CHAR:
                            //Read right away to set default battery level
                            caracteristic.Read().Subscribe(result => {
                                BatteryLevel = (int)result.Data[0];
                            });
                            await caracteristic.EnableNotifications();

                            caracteristic.WhenNotificationReceived().Subscribe(result =>
                            {
                                BatteryLevel = (int)result.Data[0];
                            });
                            break;
    
                            etcetera..

 When disconnecting I disable notifications for all characteristic with this property and then clear the list. I keep the reference to the IDevice:

foreach(IGattCharacteristic charac in characteristics)
                    {
                        if(charac.IsNotifying)
                        {
                            charac.DisableNotifications();
                            Debug.WriteLine("", "Disabling notification");
                        }
                    }
                    characteristics.Clear();
                    characsFound = 0;

I cant see that i am keeping any references to the the characteristics. Am i missing something obvious? The BLE device is used in other native apps so the sensor is not the issue.

## Actual Behavior
Number of characteristics should be the same upon reconnect.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

closed 2018-02-08 by MDLSKF

MDLSKF, 2018-02-08:
Solved..Bad reactive extensions skills was to blame.

aritchie, 2018-02-08:
I'm glad you solved it, but I see a bunch of other things that are going to give you problems:

1) Don't try to unsubscribe on disconnecting.  If you can control the "disconnect" process, do that before hand.  Most host devices are good at clearing the subscriptions on disconnect anyhow.  In your current case, most of those calls will fail.
2) If you are only using a handful of characteristics, you could use WhenKnownCharacteristicDiscovered (coming in beta3) or GetKnownCharacterisitics to streamline the detection process.
3) WhenAnyCharacteristic... will fire each time there is a new connection made to the device.  There is no need to store anything globally.  You can iterate through the list and just check if the device is connected before each loop through the characterisitics.

MDLSKF, 2018-02-15:
Got it, thanks!