aritchie/bluetoothle #132: Android Deadlock after GetKnownCharacteristics (both on 5.3.2 and 6.0.0-beta3)

agouriou, 2018-02-12:
First of all, thanks for this plugin.

## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
8.0

## Version of Library
5.3.2 and 6.0.0-beta3

## Expected Behaviour
GetKnownCharacteristics can be used.

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
After calling GetKnownCharacteristics, a deadlock occurs when trying to call EnableNotifications() on a characteristics.

Here is the code. The deadlock appears in the call to EnableNotifications.
`public IObservable<Unit> DiscoverAndSetupCharacteristicsZZ(IObservable<IGattService> service)
        {
            return service.SelectMany(s =>
            {

                // write charac is ready as soon as we find it
                IObservable<bool> writeCharacReady =
                        s.GetKnownCharacteristics(WRITE_CHARAC_UUID)
                        .Do(OnWriteCharacFound)
                        .Select(charac => true);

                // need to enable notifications on read charac before it's ready
                IObservable<bool> readCharacReady =
                        s.GetKnownCharacteristics(READ_CHARAC_UUID)
                        .Do(OnReadCharacFound)
                        .SelectMany(charac => charac.EnableNotifications());
                // sans le enable notification ça marche, sinon, seul a read characteristic est lue

                // just a return a unit to indicat that device is ready to be used
                return Observable.Zip(readCharacReady, writeCharacReady)
                            .Select(res => new Unit());
            })
                .Take(1);
        }`

 The lock is in **Plugin.BluetoothLE.Android.Internals.DeviceContext,** line 42: `reset.WaitOne()`

I've tried both v5.3.2 and 6.00-beta3 (I've seen that you've worked on this area in release 6). The deadlock is always there.

If i use WhenCharacteristicDiscovered, no deadlock appears.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-02-12:
Do the requests for known characteristics in 1 call.  GetKnownCharacteristicis(Guid, Guid);

agouriou, 2018-02-12:
Thanks for this quick answer. 

If I do it in one call, I still get a lock on `EnableNotifications` if I do it in the same subscription process.
I need to do it in different subscriptions, and I can't chain the Observable (as here with SelectMany).

This is probably the same underlying issue?

aritchie, 2018-02-13:
Chaining everything in android is going to be difficult.  The problem with android is that most things need to be done on the main thread and in sequence.  

I have an idea how to fix this, but it isn't coming quick.  The easiest fix for you is to do the following

var charater = await GetKnownCharacteristic(...); // this will release the lock
character.Write/Read()

agouriou, 2018-02-13:
Allright, thanks for the explanation!

> The problem with android is that most things need to be done on the main thread and in sequence.

Did you find this out of experiments or did you read this somewhere on the android documentation?

aritchie, 2018-02-13:
The many defects with android are known throughout the industry.  How to deal with them has come by trial and error

agouriou, 2018-02-27:
Hi.
I'm trying to implement disconnection and re-connection.
When connecting, discovering of characteristics is successfull, and`EnableNotification` also.

Disconnection works fine.

However, when doing a connection again , discovering of characteristics is successfull, but `EnableNotification` exhibits the same deadlock as previously.

The deadlock is present though I don't chain all calls. I do characteristics discovering, enabling notification and reading in different "subscriptions".

Am I missing something? Maybe on the disconnection?

Thanks

aritchie, 2018-02-27:
I don't know what your current code base looks like.  I recreate the devicecontext each time so there shouldn't be any pending locks.

ianewby, 2018-03-05:
I can confirm the same issue with reconnection (6.0.0 Beta 3)

Code sample below:

                    var btcs = await BluetoothDevice.GetKnownCharacteristics(SSServiceUuid, BTCWrite, BTCRead).Take(2).Timeout(TimeSpan.FromSeconds(5)).ToList();
                    if (btcs.Count() != 2)
                    {
                        BluetoothDevice.CancelConnection();
                        return false;
                    }
                    GCWrite = btcs.First(x => x.Uuid == BTCWrite);
                    GCRead = btcs.First(x => x.Uuid == BTCRead);
                    await GCRead.EnableNotifications();

On the first connection runs through fine. Second connection fails at GCRead.Enable Notifications().

If I connect again for a third time while this is hanging - it works and the handing thread finishes.

Should I wrap all this in a Device.BeginInvokeOnMainThread  ?

Regards
Ian Newby

aritchie, 2018-03-05:
Yes.  If you set the two flags the way I show above

aritchie, 2018-03-06:
Please try 5.3.3 - note that removing the lock from this method will cause android's defect to become apparent meaning that you can only do 1 operation at a time (sequentially).

ianewby, 2018-03-06:
I had to move from 5.3.3 as 5.3.3 struggled when trying to connect/disconnect from different devices. We are trying to connect to industrial sensors where they may be 20 in one room.

We scan, connect to each one one by one and get the data from them, then disconnect. The user can reconnect to them to update info stored in them. This is when it is failing.

With 5.3.3 the disconnect breaks subsequent connections.

What are the two flags you mention - I cant find any reference in the ticket?

Thanks for all your work!

aritchie, 2018-03-06:
@ianewby I only changed a couple of lines between 5.3.2 to 5.3.3.  None of it was around connection.  Please stay on the topic of this issue.  

As for the flags, DO NOT NOTE, if you use these, you will have to fight with android bugs.  I will not help you if these are set

```csharp
// in your android project
CrossBleAdapter.AndroidDisableLockMechanism  = true;
CrossBleAdapter.AndroidPerformActionsOnMainThread = false;
```

aritchie, 2018-03-23:
Deadlocks are solved since the lock mechanism has been turned off by default

closed 2018-03-23 by aritchie