aritchie/bluetoothle #134: Corrupted data recieved after disconnect / reconnect in iOS

MDLSKF, 2018-02-15:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS 11.2.5

## Version of Library
6.0-beta3

## Expected Behaviour
*  Use the BLE library to connect and gather known data from a BLE sensor
* Disconnect from the sensor
* Reconnect to sensor
* Gather data once again -> Data has the same previous fixed values

## Actual Behavior
*  Use the BLE library to connect and gather known data from a BLE sensor
* Disconnect from the sensor
* Reconnect to sensor
* Gather data once again -> The data is corrupted

This is not occurring on Android 7.1.1or 7.0

**Connect**

public void Connect()
        {
            //To avoid reconnecting after disconnect
            var config = new GattConnectionConfig { IsPersistent = false };

            connectDisposable = device.Connect(config).Subscribe(_ =>
            {
                    // Only subscribe once
                    if (!anyCharacteristicDiscoveredSubscribed)
                    {
                        device.WhenAnyCharacteristicDiscovered().Subscribe(async characteristic =>
                        {
                            characsFound++;
                            Debug.WriteLine(string.Empty, "Characteristics found: " + characsFound);
                            if (characteristics.Count < totalCharacteristicsCount)
                            {
                                characteristics.Add(characteristic);
                            }

                            if (characteristics.Count == totalCharacteristicsCount)
                            {
                                await SetupAllCharacteristics();
                            }
                        });
                        anyCharacteristicDiscoveredSubscribed = true;
                    }
                });
        }

**Recieve data from characteristic**

                            //Subscribe only once
                            if(!dataNotificationSubscribed)
                            {
                                characteristic.WhenNotificationReceived().Subscribe(data =>
                               {
                               //DEBUGGING
                               for(int i = 0; i<data.Data.Length; i++)
                               {
                                   byte dataB = data.Data[i];
                                   Debug.WriteLine("data byte " + i.ToString() , " " + dataB.ToString("x8"));
                               }

                                   if (dataParser != null)
                                   {
                                       dataParser .ParseDataFromBytes(data.Data);
                                   }
                               });
                                dataNotificationSubscribed= true;
                            }

**Disconnect**

        public void Disconnect()
        {
            device.CancelConnection();
        }

                   //When recieving disconnect status clear all characteristics and prepare for reconnect
                   device.WhenStatusChanged().Subscribe(status => {
                    //Disconnected
                   if (status.Equals(ConnectionStatus.Disconnected))
                   {
                       connectDisposable.Dispose();
                       characteristics.Clear();
                       characsFound = 0;
                       //Tell subscribers we are disconnected
                       connectionStatusSubject.OnNext(DeviceConnectionStatus);
                       statusChangedSubscribed = false;
                   }
                   }

**DEBUG OUTPUT BEFORE 1st DISCONNECT**
(HEX : INDEX)

9a: data byte 0  
99: data byte 1  
99: data byte 2  
3f: data byte 3 
66: data byte 4 
66: data byte 5 
66: data byte 6 
3f: data byte 7 
33: data byte 8 
93: data byte 9 
88: data byte 10 
c3: data byte 11
0: data byte 12 
0: data byte 13 
0: data byte 14 
0: data byte 15 
0: data byte 16 
0: data byte 17 
0: data byte 18 
0: data byte 19

**DEBUG OUTPUT AFTER RECONNECTING**

69: data byte 0 
10: data byte 1 
8b: data byte 2 
b: data byte 3 
d2: data byte 4 
7: data byte 5 
1: data byte 6 
b: data byte 7 
30: data byte 8 
2: data byte 9 
13: data byte 10 
3: data byte 11 
ce: data byte 12 
4: data byte 13 
d4: data byte 14 
6: data byte 15 
d: data byte 16 
24: data byte 17 
65: data byte 18 
2a: data byte 19
--

                   
## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-02-15:
Please review the source code in cases like this.  I don’t alter the response in any way and the code will support that fact. Please check other sources before sending out issues like this

closed 2018-02-15 by aritchie

MDLSKF, 2018-02-15:
[image]
[image]

I´m sorry if my description lacked details. I really cant figure out what the issue is.The only thing I notice in the call stack is that the received data values changes after receiving the CBCharacteristicEventArgs in GattCharacteristic. The first image shows the correct data, the second the corrupted.

aritchie, 2018-02-15:
I think you're missing the point of my message.  If I don't touch the bytes coming from the characteristics at all.... I'm not sure how this is an issue with this library!?  I also can't tell if your message is corrupt or not since I don't know the format of the packet.  Please think about the types of things you are filing and the answers you're hoping for.  

I get that you are having trouble figuring stuff out... but I can't do support beyond issues with the library.    You are welcome to debug using the raw source code.  I would suggest checking google as it could be an issue with iOS or your host device - I honestly don't know.