aritchie/bluetoothle #136: BluetoothLECharacteristic.OnReadString is called three times in beta3

mt104, 2018-02-17:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

iOS 11.2.5

## Version of Library

6.0.0-beta3

## Expected Behaviour

Calling this code once...

            _characteristic.Read().Subscribe(result =>
            {
                handler(result.Data);
            });

Should result in this code being called once on other device...

                BluetoothLECharacteristic cMessage0 = service.AddCharacteristic(_sCharacteristicUUIDMessage0);
                cMessage0.OnReadString += () =>
                {
                    string sArticle = _returnMessage(0);
                    return sArticle;
                };

## Actual Behavior

OnReadString() is actually called 3 times. On each occasion I return the correct data. The data that makes it back to the other device is 3 copies of the data appended.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-02-17:
First, I don’t have a function called OnReadString.  The reason you are likely getting the callback 3 times is because you are pinning the characteristics somewhere.  They will be recreated with every (re)connection

aritchie, 2018-02-21:
No response - closing

closed 2018-02-21 by aritchie