aritchie/bluetoothle #139: [Question] Initial "disconnected" status when connecting to a device on Android

agouriou, 2018-02-23:
Hi,

I'm opening this as a question since this doesn't seem to be an issue, but rather a strange behavior from Android.

When I connect to a device on Android, using `IsPersistent = false`, and I subscribe to `WhenStatusChanged()`, I get a `connected` status as expected, but right after I get a `disconnected` state. However, the connection still seems "connected", I can read/write to characteristics.

This behavior seems strange and I can see in https://github.com/aritchie/bluetoothle/blob/master/Plugin.BluetoothLE.Android/Device.cs that you take care of this when creating `CreateAutoReconnectSubscription`, you do a `Skip(1)`.

Can you confirm that this is a "normal" (though strange) behavior from Android? If I use a connection with `IsPersistent = false`, I should then skip the first "disconnected"? But how to detect when the connection will be disconnected for real?

Thanks

aritchie, 2018-02-23:
I fire most observables with current state the second you subscribe.  It is by design.

Please don’t spam in closed issues in the future

agouriou, 2018-02-23:
Thanks for your answer.

What I get is "connected", then "disconnected".
What I would expect is : "disconnected" then "connected". (this is what I can see on iOS)

aritchie, 2018-02-23:
That’s not how your title read.  You need to provide a reproducible sample then.  Please always follow the issue template as well.

agouriou, 2018-02-27:
###### Actual behavior:

`
device.WhenStatusChanged().Subscribe(status => Console.WriteLine(status));
device.Connect().Subscribe();
`
This prints :
> disconnected
> connected

`
device.Connect().Subscribe();
device.WhenStatusChanged().Subscribe(status => Console.WriteLine(status));
`
This prints :
> connected
> disconnected

###### Expected behavior:
Both case should print connected as the last statement

aritchie, 2018-02-27:
Case 1 is correct.
Case 2 I have not seen which is why I requested the full issue template and a reproducible sample.  The issue template identifies if I'm fighting with a specific device or manufacturer (which is common) or user code.

aritchie, 2018-03-31:
never received followup - closing

closed 2018-03-31 by aritchie