aritchie/bluetoothle #143: How to match between a nuget version and a commit ?

agouriou, 2018-03-01:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [X] Android
 - [X] tvOS
 - [X] macOS
 - [X] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

## Version of Library

## Expected Behaviour
A git tag for each nugget version would be great.

## Actual Behavior
Is there a way to know which commit matches a nugget version?
For example, which commit corresponds to nugget 5.2.2?

Thanks

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-03-01:
You don't.  This project was written for my needs originally and I was iterative in updating it.  

Please only use this for feature requests & defects in the library in the future.

closed 2018-03-01 by aritchie

agouriou, 2018-03-01:
Thanks. Which process should I use to ask questions?

aritchie, 2018-03-01:
I'm sorry, I'm not really here for support.  I don't have time to do day to day on these things.  It is too much of a chore with the amount of questions I get in day seeing as I don't make a cent off it.  I write the code, people choose to use it or not.  

That said, I do attempt to answer questions where the user has at least attempted to do their homework before submitting something.  Issues & feature requests I always make an effort to do something about.