aritchie/bluetoothle #146: Why does the write command give "echo" result?

biapar, 2018-03-04:
Hi,

I use your lib into Xamarin Forms .NETStandard 2.0 and I'm able to scan and connect to devices and scanning characteristics. I've HM-10 module to emule serial port. I've to read / write on FFE1 characteristic on service FFE0.
The problem is that I've the "echo" command into result. That is: I send "CHKST" and into result I've "CHKST".

The code is:

 ```
public static Guid SSServiceUuid = Library.GuidFromUuid("FFE0");
public static Guid SSCharacteristicUuid = Library.GuidFromUuid("FFE1");

public static Guid GuidFromUuid(this string uuid)
        {   // 00002a28-0000-1000-8000-00805f9b34fb 
            //this sometimes returns only the significant bits, e.g.
            //180d or whatever. so we need to add the full string
            var id = uuid.ToString();
            if (id.Length == 4)
            {
                id = "0000" + id + "-0000-1000-8000-00805f9b34fb";
            }
           
            return Guid.ParseExact(id, "d");
        }

```
The code where I try to write:

 ```
BluetoothDevice.WhenAnyCharacteristicDiscovered().Subscribe(async characteristic =>
                    {

                        IGattCharacteristic c = characteristic;
                        IGattDescriptor d;

                        CharacteristicsList.Add(c);
                        Console.WriteLine("Characteristic: " + c.Uuid + " Description:" + c.Description.ToString() + " Propr: " + c.Properties.ToString() );

                        if (c.Uuid== SSCharacteristicUuid)
                    {
                            Console.WriteLine("Servizio / Caratteristica Generica trovata");
                            Console.WriteLine("" + c.Service);

                            var result = await c.Write(Encoding.UTF8.GetBytes("CHKST\n")).Timeout(TimeSpan.FromSeconds(5)).ToTask();
                            Console.WriteLine("Risposta BT write in scopri characts: " + Encoding.UTF8.GetString(result.Data, 0, result.Data.Length));

                        }

                    });

```
Why? Could you give some help, please?
Thanks.

aritchie, 2018-03-04:
This is not a general support site.  I will respond to defects and feature requests to do with this library as long as the issue template has been filled out (the one you deleted before posting).  Seeing as though you are relatively new to OSS, I’ll point you to two things

1) serial protocols generally echo like what you are seeing.  You should check you host device documentation 

2) A’s stated in the faq section of the docs for this library, you need to be careful how storing references to services, characteristics, and descriptors

closed 2018-03-04 by aritchie

biapar, 2018-03-04:
Hi,

1. I know serial protocol and I suppose that who build the card has not configured something on serial port and on HM10. I tried with LightBlue and BLE Scanner apps.
2. I'm not storing anything at the moment. I made all into subscribe for testing.