aritchie/bluetoothle #147: Deadlock write Description

Nebudef, 2018-03-05:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
7.0

## Version of Library
5.3.2 

## Expected Behaviour
Descriptor will be written

## Actual Behavior
Deadlock, nothing happen

```
Guid beacon = Guid.Parse("00000000-0000-0000-0000-eb806e44cfbc");
IDevice device = CrossBleAdapter.Current.GetKnownDevice(beacon);
await device.Connect();
var service = await device.GetKnownService(SHT_SENSOR_SERVICE_UUID);
var characteristic = await service.GetKnownCharacteristics(SHT_SENSOR_CHARACTERISTIC_DATA_SYNC);
await characteristic.EnableNotifications(true);
characteristic.RegisterAndNotify().Subscribe(result =>
{
              //I Never Reach this
              Debug.Print("GetaNotifcation");
});
 characteristic.WhenDescriptorDiscovered().Subscribe(async result =>
{
                //This will be never Finished
                await result.Write(new Byte[] { 0x01, 0x00 });
 });`
```

Did i missused somthing? 
or Do i generally made a mistake?

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

mullerch, 2018-03-05:
Regarding this issue : https://github.com/aritchie/bluetoothle/issues/137#issuecomment-368223714

You should call RegisterAndNotify in the WhenDescriptorDiscovered callback. And you should not use both RegisterAndNotify and EnableNotifications.

That said, I also have a similar problem, calling RegisterAndNotify hangs. The last log I get is :

> D/BluetoothGatt(16033): setCharacteristicNotification() - uuid: c1b25013-caaf-6d0e-4c33-7dae30052840 enable: true

@aritchie The documentation seems a little bit outdated and it seems that a multiple questions could be answered by this doc. Also it is very hard to find information about the BluetoothLE process of connecting-discovering-subscribing that seems problematic in Android. In the related issue, you say "You really should take a look at what each of these methods do". I see no XML comment in the library and no "formal" description of the API in the doc [here](https://github.com/aritchie/bluetoothle/tree/master/docs), am I missing something ? In the meantime, could you provide an example of usage in this scenario that could help us to make this more straightforward ? I did of course look at the sample program but the process is not "automatic", all steps are started by user actions. Thanks for your library and the quick answer time you provide !

mullerch, 2018-03-05:
Apparently, I had to do the RegisterAndNotify on the UI thread.

Nebudef, 2018-03-05:
I have Played a littel with the Code but my Problem stays the same.
It is not so that i could not get a Notification, i catch exactly one aslong i do not await on my write call.
But as soon as i use await there it never get Finished.

```
IDevice device = CrossBleAdapter.Current.GetKnownDevice(beacon);
await device.Connect();
var service = await device.GetKnownService(SHT_SENSOR_SERVICE_UUID);
var characteristic = await service.GetKnownCharacteristics(SHT_SENSOR_CHARACTERISTIC_DATA_SYNC);
characteristic.WhenNotificationReceived().Subscribe(result2 =>
{
         Debug.Print("Gat a Notification");
 });

 characteristic.WhenDescriptorDiscovered().Subscribe(async result =>
{
                characteristic.RegisterAndNotify(true).Subscribe();
                await result.Write(new byte[] { 0x01, 0x00 });
                //Never Reached
                Debug.WriteLine("Sucess");
 });
```

aritchie, 2018-03-05:
@Nebudef I have attempted to deal with android ble stacks in so many different ways and nothing is ever quite right.  You can disable my attempts to deal with everything being called on the main thread and being forced sequentially by doinh the following (note that you now deal with all of the issues yourself):

```csharp
// can only be set from android project
CrossBleAdapter.AndroidDisableLockMechanism = true;
CrossBleAdapter.AndroidPerformActionsOnMainThread = false;
```

or you can wait (possibly forever) until I figure out a solution for all of the shit that is android ble dev

aritchie, 2018-03-05:
@mullerch I don't really appreciate the overall tone of your comment, nor the fact that you side tracked this issue with requests for docs, but I'll answer in any case.

> The documentation seems a little bit outdated and it seems that a multiple questions could be answered by this doc

You're right.  It is.  Not enough time in the day anymore.  It often doesn't matter how much I document, people will still ask the repeated questions, so it has burned me out of taking time to do this.

> Also it is very hard to find information about the BluetoothLE process of connecting-discovering-subscribing that seems problematic in Android.

Google it.  You'll find all of the fires I'm trying to fight. 

> In the related issue, you say "You really should take a look at what each of these methods do". I see no XML comment in the library and no "formal" description of the API in the doc here, am I missing something ?

The source code is all open.  You should look at it when docs aren't present.  If you are upset with the fact that I haven't done it, why not contribute back?

> In the meantime, could you provide an example of usage in this scenario that could help us to make this more straightforward 

No - for several reasons.  

- The  main one being that I'm not looking for library adoption.  If people find value here - great.  I hope I can continue to improve it.  
- If I don't have the time to update all of the docs, what makes you think I have time to do another sample?  
- what is straight forward to some, is not to others
- The fact is that I build this for my needs.  You are welcome to contribute something more straight forward if you want.  It would be appreciated.

aritchie, 2018-03-06:
Please give 5.3.3 a try - I don't have access to my testing device this week so I need feedback please

mullerch, 2018-03-06:
@aritchie Sorry if I was rude, I didn't mean to be. Doc is always a pain to maintain and I understand the time argument. It was just in case you didn't realize it was outdated (as you're probably not using it). Looking at the sources is a good point and I will for sure contribute (as you may notice) if possible.

Regarding the sample, I didn't expect you to give time for this. I was just asking if you were ok to share some code of your own projects using the library, more as an informal doc than a fully working project.

And sorry for side tracking the issue...

Nebudef, 2018-03-07:
@aritchie sorry for the late feedback, it took me a while to test it
the first workaround 
```
CrossBleAdapter.AndroidDisableLockMechanism = true;
CrossBleAdapter.AndroidPerformActionsOnMainThread = false;
```
worked. I get no deadlock anymore and i get the notifications i expected.
In my short code i get no Errors so far.

With 5.3.2 with no workaround i get one Notification and the Deadlock.
With 5.3.3 with no workaround i get still the Deadlock and no notification.
With 5.3.3 and workaround i get all notifcations as expected.

aritchie, 2018-03-07:
@Mullerch I appreciate it.  I will try to get some sort of sample update in the near future.  The unit tests also have some more “direct to device” samples

@nebudef thanks for the update and tests.  I will continue to work through the issue.  Keep in mind, while those flags may appear as a workaround, you are now exposed to the wonderful world of android defects.

cbfrank, 2018-03-15:
I have the similar issue. My code is below:
`
var characteristicReturn = await GetCharacteristic(serviceUUID, characteristicReturnUUID,
                c => (c.Properties & CharacteristicProperties.Notify) == CharacteristicProperties.Notify);
            var characteristicRequest = await GetCharacteristic(serviceUUID, characteristicRequestUUID,
                c => (c.Properties & CharacteristicProperties.Write) == CharacteristicProperties.Write);
            var receivedNotifyResults = new List<CharacteristicResult>();
            try
            {
                var results = await characteristicReturn.RegisterAndNotify().Do(r => receivedNotifyResults.Add(r)).TakeWhile(r => somecode)
                    .Merge(characteristicRequest.Write(data).FirstAsync()).Timeout(TimeSpan.FromSeconds(30)).ToArray();
                //ToDo: Check characteristicNotify is notify off
                return receivedNotifyResults.ToArray();
            }
            catch (TimeoutException e)
            {
                
                return receivedNotifyResults.ToArray();
            }
`
This code works with 5.3.3, but after upgrade to 5.3.4, the line var results = await characteristicReturn... will never returned, and will get timeout exception

aritchie, 2018-03-15:
@cbfrank Please check the release notes for the flags I swapped off.  You're now responsible for the main thread hops and sequential hits to the device on Android.  There were too many deadlocking issues going on because of my stuff.  Now, they are happening because of Android.  You can turn the flags on, but don't file a defect if they cause you grief, I'm well aware.  Thanks

Nebudef, 2018-03-22:
@aritchie while having another Issue in my code i have update to 5.3.4 and it semmes that the Deadlock do not happen anymore

closed 2018-03-23 by aritchie