aritchie/bluetoothle #148: device status subscriptions eventually fail after reconnect

x37v, 2018-03-05:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue

7.0 and 8.1.0

## Version of Library

master and 6.0.0-beta3

I haven't been able to successfully build the dev sample project for android.

## Expected Behaviour

**Sample.Droid**

`device.WhenStatusChanged()` should fire everytime status changes.

## Actual Behavior

I have a peripheral device that has a button to kill the bluetooth connection, I use the **Sample.Droid** app to scan for my device then connect, then I kill the connection from my bluetooth peripheral device, the ble library reconnects but usually after the first cycles of killing the bluetooth connection and reconnecting the library fails to call `device.WhenStatusChanged()`, DeviceViewModel.cs line 135. I verified with a breakpoint but also the GUI doesn't update the connection status.
the `Ble.Adapter.WhenDeviceStatusChanged()` subscription does get called, and I see:
`[BluetoothGatt] onConnectionUpdated()`
in the application output log.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

x37v, 2018-03-05:
the `BleAdapter.WhenDeviceStatusChanged()` also often triggers, incorrectly indicating that a device is connected initially on disconnect before an automatic reconnect.

aritchie, 2018-03-05:
what status(es) is triggering incorrectly?  Is this a specific set of devices?

x37v, 2018-03-05:
@aritchie the status give is `Connected` on disconnect. I've tested this with:

* samsung galaxy note 5 on android 7.0
* huawei nexus 6p on android 8.1.0

both fail to give status messages from the `device.WhenStatusChanged()` after a reconnect, both eventually give `Connected` status on disconnect in the `BleAdapter.WhenDeviceStatusChanged()` subscription

aritchie, 2018-04-14:
Fixed in v6

closed 2018-04-14 by aritchie