aritchie/bluetoothle #151: iOS GattCharacteristic ArgumentException message

Zaibatsu89, 2018-03-06:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS all versions

## Version of Library
5.3.3

## Expected Behaviour
An ArgumentException with message: "15". When the attribute requires encryption before it can be read or written. See also: [CBATTErrorExtensions Class - Xamarin](https://developer.xamarin.com/api/type/CoreBluetooth.CBATTErrorExtensions/), [CBATTError Enumeration - Xamarin](https://developer.xamarin.com/api/type/CoreBluetooth.CBATTError/) and the Android equivalent [BluetoothGatt | Android Developers](https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#GATT_INSUFFICIENT_ENCRYPTION).

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_

An ArgumentException with message: "Encryption is insufficient.".

## Suggested change
String messages are prone to changes. A `CBATTError` enumeration has constant values. The `IObservable` function `OnError` requires an `Exception`, so using this enumeration as his `message` satisfies.

Beware: I did not test the following syntax.

In: Plugin.BluetoothLE.Apple.Shared/GattCharacteristic.cs, lines 46, 85 and 131.
From: `ob.OnError(new ArgumentException(args.Error.ToString()));`
To: `ob.OnError(new ArgumentException(CBATTErrorExtensions.GetDomain(args.Error)));`
Or: `ob.OnError(new ArgumentException(args.Error.Description));`

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-03-06:
I'd take a tested PR for this :)

jeroenwitteveen, 2018-03-09:
Great enhancement proposal, error strings are not reliable indeed! Beside the error text message please add an error number. It would be even better if the plug in would translate the differences in error numbers between the different platforms (Android, iOS, ...) to a general plugin error number!

commit 7cc0050b (2018-04-01): Fix #151 - more meaningful messages on ios
closed 2018-04-01 by aritchie