aritchie/bluetoothle #152: RegisterAndNotify does not subscribe after device disconnect

mullerch, 2018-03-07:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
Android 4.4.2
SM-T230

## Version of Library
5.3.3 (NuGet)

## Configuration
Connect
- AutoConnect = true
- IsPersistent = false

Problem occurs with the following code :
`IGattCharacteristic.RegisterAndNotify().Subscribe(CharacteristicOnValueUpdated);`

Does not occur with :
`
IGattCharacteristic.EnableNotifications().Subscribe();
IGattCharacteristic.WhenNotificationReceived().Subscribe(CharacteristicOnValueUpdated);
`

The problem occurs when the devices disconnects, but not when the connection is closed with IDevice.CancelConnection().
Only the IDevice handler is kept after connection loss.

## Expected Behaviour
1. Discovery of the device
2. Connection to the device
3. The device connects
4. Subscription to a characteristic with RegisterAndNotify 
5. The characteristic is subscribed and the callback is called when there is a new value
6. Disconnection of the device, from the device side (link loss, or device goes in sleep)
7. Devices comes in range
8. Re-connection to the device, the WhenStatusChanged is called
9. Subscription to a characteristic with RegisterAndNotify 
10. The characteristic is subscribed and the callback is called when there is a new value

## Actual Behavior
(1-9 is the same than above)
1. Discovery of the device
2. Connection to the device
3. The device connects
4. Subscription to a characteristic with RegisterAndNotify 
5. The characteristic is subscribed and the callback is called when there is a new value
6. Disconnection of the device, from the device side (link loss, or device goes in sleep)
7. Devices comes in range
8. Re-connection to the device, the WhenStatusChanged is called
9. Subscription to a characteristic with RegisterAndNotify 
10. The characteristic is not subscribed (also not for the device)

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-03-07:
This sounds like it is by design.  Characteristic discovery must occur for every (re)connection and thus, all subscriptions must be recreated as a result.

Just to be sure I understand you steps, please include a code snippet

mullerch, 2018-03-07:
Characteristic discovery is correctly done after re-connection. I then try to recreate the subscription.

https://pastebin.com/Pd4B1YBm

aritchie, 2018-03-23:
Could not reproduce.

closed 2018-03-23 by aritchie