aritchie/bluetoothle #153: AD Service data is reversed in iOS

mullerch, 2018-03-13:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iOS 11.2

## Version of Library
5.3.4

## Expected Behaviour
IScanResult.AdvertisementData.ServiceData[0] contains the first service data field defined in the ADV packet

## Actual Behavior
IScanResult.AdvertisementData.ServiceData[1] contains the last service data field defined in the ADV packet. Actually I've tested with an ADV with two service data fields only, so maybe it's the second and not the last, but the last makes more sense.

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

mullerch, 2018-03-13:
Behavior on Android is actually correct. Therefore abstraction is problematic.

aritchie, 2018-03-23:
I don't see an issue with what you're describing here.  I'm not doing anything to reverse the order of the packets here.  Please send a PR if you feel otherwise.

closed 2018-03-23 by aritchie