aritchie/bluetoothle #155: iOS WriteWithoutResponse crashing

borbmizzet, 2018-03-19:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [*] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
iPhone 7 10.3.3

## Version of Library
6.0.0-beta3

## Expected Behaviour
Not crashing when awaiting Characteristic.WriteWithoutResponse(byte[]])

## Actual Behavior
_If applicable, please include entire exception - type, message, stacktrace_
App crashes when calling WriteWithoutResponse on iOS.

DebugLog:
-[CBPeripheral canSendWriteWithoutResponse]: unrecognized selector sent to instance 0x1746e1380

Stack Trace:
CoreFoundation: __exceptionPreprocess
libobjc.A.dylib: objc_exception_throw
CoreFoundation: -[NSObject(NSObject) doesNotRecognizeSelector:]
CoreFoundation: ___forwarding___
CoreFoundation: _CF_forwarding_prep_0
MyApp.iOS: wrapper_managed_to_native_ObjCRuntime_Messaging_bool_objc_msgSend_intptr_intptr <unknown>:1
MyApp.iOS: CBPeripheral.g.cs:263
MyApp.iOS: Plugin_BluetoothLE_GattCharacteristic__c__DisplayClass11_0__WriteWithoutResponseb__0_System_IObserver_1_Plugin_BluetoothLE_CharacteristicGattResult <unknown>:1

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-03-19:
Native crashes have nothing to do with me.  You need to talk to xamarin for stuff like this!

borbmizzet, 2018-03-19:
@aritchie According to [this documentation](https://developer.apple.com/documentation/corebluetooth/cbperipheral/2891512-cansendwritewithoutresponse?language=objc), CBPeripheral.cansendwithoutresponse is only compatible with iOS11+.  This is mirrored in the Xamarin source code [here](https://github.com/xamarin/xamarin-macios/search?utf8=%E2%9C%93&q=canSendWriteWithoutResponse&type=)  Your readme.md says that your plugin supports iOS 7+, yet you use CBPeripheral.cansendwithoutresponse at Plugin.BluetoothLE/Platforms/Apple/GattCharacteristic.cs:WriteWithoutResponse:42.
This isn't a Xamarin problem, I'm sorry to say.  It is an oversight in your library.  Whether you choose to bump your minimum req's or find a workaround that doesn't use canSendwithoutResponse is up to you, though I'd hope for the latter.

aritchie, 2018-03-19:
Listen - I'll fix this, but PR's only on betas from now on!  Otherwise, stick to the release.

borbmizzet, 2018-03-19:
Much appreciated.  But if I may ask, if you don't want bug reports on betas, doesn't that defeat the purpose of releasing betas?  I've got no problem respecting your wishes here, but I always thought the whole point of releasing a beta was so people could find bugs and report them so they can be fixed prior to releasing a "production-ready" version.

commit 18d6e3c1 (2018-04-01): Fix #155 - ensure ios compat <11
closed 2018-04-01 by aritchie