aritchie/bluetoothle #157: IOS issue

Rubnyk, 2018-03-20:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
IOS 11

## Version of Library
5.3.4

## Expected Behaviour
1. Connect to device
2. read some GhatCharactristic like Acceleromter, temperature
3. CancelConnection
4.Connect again
5. read some GhatCharactristic like Acceleromter, temperature

expected to see the value changes like the first time it connected

## Actual Behavior
1. Connect to device
2. read some GhatCharactristic like Acceleromter, temperature
3. CancelConnection
4.Connect again
5. read some GhatCharactristic like Acceleromter, temperature

Now it sending the  notification  with the same value over and over

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-03-20:
This likely has something to do with your host device.  There is nothing I do to replay read values.  

Please include actual details in your filings.  I'm not psychic - AGAIN - as the issue template says.

Rubnyk, 2018-03-21:
It happens on IOS only! in android it work fine . i get the value ok so i don't think it's a problem with the hardware,
what more details do you need?

aritchie, 2018-03-21:
I need details like a small reproducible sample of code.  I can’t offer any type of support if I can’t see a problem.

Rubnyk, 2018-03-21:
here is a link to my full code (i put it in my one drive)

https://1drv.ms/u/s!AiWji7XXvfBfiKh5IDvYTXg66XcXag

aritchie, 2018-03-21:
Please -  a small reproducible sample.  I'm not debugging 600 lines of code, nor do I have a device that matches this profile.  I also have no details as to what is giving you "same value over and over again".  This is not a support department.  You have to basically serve up the problem as small and as quick as possible if you want help, otherwise you'll find you'll get no help from anyone is the OSS community.

Now - waiting on a smaller sample or at least a ton more details.  I need the following info:
What characteristic is giving you the same value over and over again?  
How do you know the value has changed on the host device?  
Have you tried with the sample app - does it reproduce there as well?

closed 2018-03-23 by aritchie