aritchie/bluetoothle #159: Notifications Not Enabled on Android

heiser, 2018-03-22:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [X] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
7.0

## Version of Library
v5.3.4

## Expected Behaviour
When building my Xamarin Forms app for iOS, it all works fine. The same code for Android is not working. The application scans for available devices, and when it finds one that offers the specified services, it connects, discovers characteristics and wires up the notifications. Once that is done, we have the peripheral send data on the notify characteristic, which the app should receive. On iOS, it does.

## Actual Behavior
On Android, the same code does not work correctly. It finds the peripheral and discovers the characteristics correctly. However, when the peripheral attempts to send data on the notify characteristic, it receives an error indicating the notify characteristic has not been enabled. I do call EnableNotifications() on the characteristic, and as mentioned, it does work on iOS. No exceptions are thrown during runtime.

```
            // Scan for a device supporting the services we need
            var scanner = CrossBleAdapter.Current.Scan(
                new ScanConfig
                {
                    ServiceUuids = new List<Guid>(new Guid[] {
                        appControlServiceId,
                        serverControlServiceId
                    })
                }
            )
            .Subscribe(scanResult =>
            {
                if (_device != null)
                    return;

                // We found a device, so let's connect to it
                IDevice device = _device = scanResult.Device;
                
                // Wire up the device connection handler...
                device.Connect().Subscribe(conn =>
                {
                    Debug.WriteLine("DeviceInterface.AttemptConnection: Peripheral found. Name = " + device.Name);

                    // Discover the characteristics that we want
                    device.WhenAnyCharacteristicDiscovered().Subscribe(characteristic =>
                    {
                        // Is it the mobile app data out characteristic
                        if (characteristic.Uuid == mobileAppDataOutCharacteristicId && !_mobileDataOutDiscovered)
                        {
                            characteristic.EnableNotifications();
                            _mobileDataOutDiscovered = true;
                            _mobileDataOutCharacteristic = characteristic;

                            characteristic.WhenNotificationReceived().Subscribe(result =>
                            {
                                Debug.WriteLine("Received MobileAppDataOutCharacteristic");
                                OnMobileAppDataReceived(result);
                            });
                        }
                    });
                });
            });
        }
```

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-03-22:
Stop scanning before connecting to any device.  Android doesn’t do well otherwise

heiser, 2018-03-23:
Thank you for the quick response.

I put in code to stop scanning right before the connection. Unfortunately, the problem still occurs:

```
            // Scan for a device supporting the services we need
            _scanner = CrossBleAdapter.Current.Scan(
                new ScanConfig
                {
                    ServiceUuids = new List<Guid>(new Guid[] {
                        appControlServiceId,
                        serverControlServiceId
                    })
                }
            )
            .Subscribe(scanResult =>
            {
                if (_device != null)
                    return;

                // We found a device, so let's connect to it
                IDevice device = _device = scanResult.Device;

                // We found a matching device so stop scanning
                _scanner.Dispose();
                _scanner = null;

                Debug.WriteLine("Stopped the scan because we're about to connect...");
                Thread.Sleep(1000);

                // Wire up the device connection handler...
                device.Connect().Subscribe(conn =>
                {
                    Debug.WriteLine("DeviceInterface.AttemptConnection: Peripheral found. Name = " + device.Name);

                    // Discover the characteristics that we want
                    device.WhenAnyCharacteristicDiscovered().Subscribe(characteristic =>
                    {
                        // Is it the mobile app data out characteristic
                        if (characteristic.Uuid == mobileAppDataOutCharacteristicId && !_mobileDataOutDiscovered)
                        {
                            Debug.WriteLine("DeviceInterface.OnDeviceAdapterChange: Subscribing to MobileAppDataOutCharacteristic");

                            characteristic.EnableNotifications();
                            _mobileDataOutDiscovered = true;
                            _mobileDataOutCharacteristic = characteristic;

                            characteristic.WhenNotificationReceived().Subscribe(result =>
                            {
                                Debug.WriteLine("Received MobileAppDataOutCharacteristic");
                                OnMobileAppDataReceived(result);
                            });
                        }
```

aritchie, 2018-03-23:
You also aren't waiting until EnableNotifications is complete on the characteristic

pdonovan, 2018-03-23:
I'm not the OP, but I'm having the same problem on Android 6.0.

FYI the "Client Basics" section of the [readme](https://github.com/aritchie/bluetoothle#how-to-use---client-basics) doesn't show waiting until EnableNotifications is complete. I presume you mean doing this:

    if (characteristic.Uuid == Guid.Parse(TEMP_CHARACTERISTIC_UUID) && !_tempDiscovered)
    {
        _tempDiscovered = true;
        _tempCharacteristic = characteristic;

        characteristic.EnableNotifications().Subscribe(n =>
        {
            if (n == true)
            {
                _tempCharacteristic.WhenNotificationReceived().Subscribe(result =>
                {
                    Debug.WriteLine("Got measurement");
                });
            }
            else
            {
                Debug.WriteLine("EnableNotifications failed");
            }
        });
    }

The code above DOES work on iOS 10.3.3. On Android, `WhenNotificationReceived` is never called. I've attached a log when running on the Android 6.0 device.
[bluetoothletest.txt](https://github.com/aritchie/bluetoothle/files/1840269/bluetoothletest.txt)

heiser, 2018-03-23:
After waiting for EnableNotifications to complete, I get it to work...sometimes.

The problem seems to be that I have two notify characteristics that I am attempting to enable. When I try to enable them both, neither call to EnableNotifications complete. When I comment out one of the EnableNotification attempts, it works. It appears that they are blocking each other.

What is the recommended way to address this?

aritchie, 2018-03-23:
Android has issues!  It isn't thread safe.  It doesn't like being flooded with requests and most things need to be done on the main thread during writes (I'm going to turn the main thread stuff on again by default, but not the locking mechanism which prevents the flooding).

You need to do things sequentially with BLE on Android.  I have and am continuing to try to figure out a good way to solve this issue in the library.

closed 2018-03-27 by aritchie

heiser, 2018-03-28:
I rearranged the code to enable the characteristics sequentially, and it still does not work.

aritchie, 2018-03-28:
@Heiser I’m not psychic.  I don’t know how to help you because I can’t see any errors or code.  You have to apply additional info.

heiser, 2018-03-28:
Don't worry about it. I switched to a different BLE library and it is working now.

pdonovan, 2018-03-28:
In my case I had misread/forgotten the specs of the device I was working with, and I needed to use an Indicate `EnableNotifications(useIndicationIfAvailable: true)` to be correct. I guess iOS doesn't particularly care about the method but Android does. I wonder if the `useIndicationIfAvailable` argument should default to true?

Abdulsametileri, 2019-03-18:
` scanResult.Device.WhenAnyCharacteristicDiscovered().Subscribe(chs => {
   if (chs.Uuid == new Guid("0000FFE1-0000-1000-8000-00805F9B34FB"))
    {
        chs.EnableNotifications().Subscribe(
            result =>
            {
                Debug.WriteLine(result.Data);
                chs.WhenNotificationReceived().Subscribe(res => {
                    string ret = System.Text.Encoding.UTF8.GetString(res.Data);
                    Device.BeginInvokeOnMainThread(() =>
                    {
                        lblUvIndexVal.Text = ret;
                    });
                    Debug.WriteLine(ret);
                });
            }
        );
    }`

I solved this by subscribing EnableNotifications. Thanks for the library :)