aritchie/bluetoothle #160: dispose a scanning and reading a characteristic

Nebudef, 2018-03-27:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [] iOS
 - [x] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
7.0

## Version of Library
5.3.5

## Expected Behaviour
I work with multiple Instances of the Class who capsels my Bluetooth actions.
They use different scan callbacks. 
I now want to suppose some of them and read a Characteristic.

## Actual Behavior
`var characteristic = await device.GetKnownCharacteristics(serviceGuid, characertisticGuid);`
Gives: Object reference not set to an instance of an object.

```
private IAdapter adapter = CrossBleAdapter.Current;
private IDisposable scanner;
```
```
public void startScan(callbackBluetoothScanResult callbackScanresult, List<Guid> guids)
        {
            if (scanner != null) { scanner.Dispose(); }
            if (adapter.Status == AdapterStatus.PoweredOff)
            {
                return;
            }
            scanner = CrossBleAdapter.Current.Scan().Subscribe(result =>
            {
                if (guids.Contains(result.Device.Uuid))
                {
                    dbaTempLog01.Connection.ScanResult scanresult = new dbaTempLog01.Connection.ScanResult(result.Rssi, result.Device.Name, result.Device.Uuid, result.AdvertisementData.ManufacturerData);
                    callbackScanresult(scanresult);
                }
            });
        }
```
```
public void stopScan()
        {
            if (scanner != null) { scanner.Dispose(); }
            Debug.Print("Stop Scanning");
        }
```

```
private async Task<Byte[]> readCharac(Guid deviceGuid, Guid serviceGuid, Guid characertisticGuid)
        {
            try
            {
                var device = CrossBleAdapter.Current.GetKnownDevice(deviceGuid);
                var characteristic = await device.GetKnownCharacteristics(serviceGuid, characertisticGuid);
                var result = await characteristic.Read();
                return result.Data;
            }
            catch (Exception e)
            {
                throw e;
            }
        }
```

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-03-27:
I have no idea what you're trying to do.  I always need the full exception please

Nebudef, 2018-03-28:
First I start a task who

Nebudef, 2018-03-28:
Error ein wenig Einzugrenzen. Es scheint das er auftritt sobald ein Thread bluetooth operationen wie Schreiben und Lesen durchführt und ein anderer verucht sich für einen neuen Scan Vorgang zu listen.
Reicht das mit den bereits genannten Code Fragmenten oder wird noch mehr benötigt?
Die Fehler Nachricht ist:
System.NullReferenceException: Object reference not set to an instance of an object.
  at Plugin.BluetoothLE.Device.GetKnownService (System.Guid serviceUuid) [0x00019] Bearbeiten

I tried to narrow down the cause of the error a bit. It seems that it occurs as soon as one thread performs bluetooth operations like writing and reading and another one tries to list for a new scan process.
Is that enough with the already mentioned code fragments or is more needed?
The error message is:
System.NullReferenceException: Object reference not set to an instance of an object.
  at Plugin.BluetoothLE.Device.GetKnownService (System.Guid serviceUuid) [0x00019]

aritchie, 2018-03-28:
No. It looks as though you’ve chopped off part of the error message.  I also can’t see the code flow above.  You need to submit a reproducible sample for this.

Nebudef, 2018-03-28:
With this simplefied Code i can reproduce the Error:
```
        private void errorReproduction()
        {
            BluetoothLE bluetooth = new BluetoothLE();
            BluetoothLE bluetooth2 = new BluetoothLE();
            BluetoothLE bluetooth3 = new BluetoothLE();
            Guid guid = new Guid("00000000-0000-0000-0000-eb806e44cfbc");
            Guid JAALEE_BEACON_SERVICE = Guid.Parse("0000fff0-0000-1000-8000-00805f9b34fb");
            Guid BEACON_KEEP_CONNECT_CHAR = Guid.Parse("0000fff1-0000-1000-8000-00805f9b34fb");
            
            var task2 = new Task(() =>
            {
                TimeSpan span = TimeSpan.FromSeconds(5);
                bluetooth2.startScan(errorReproductionCallback, new List<Guid> { new Guid("00000000-0000-0000-0000-eb806e44cfbc"), }, span);
            });
            var task = new Task(async () =>
            {
                while (true)
                {
                    await bluetooth.connectDevice(guid, errorReproductionCallback);
                    await bluetooth.writeCharacteristic(guid, JAALEE_BEACON_SERVICE, BEACON_KEEP_CONNECT_CHAR, StringToByteArray("666666"));
                    await bluetooth.readCharacteristic(guid, JAALEE_BEACON_SERVICE, BEACON_KEEP_CONNECT_CHAR);
                    bluetooth.disconnectDevice(guid);
                }
            });
            task.Start();
            task2.Start();
        }
        private void errorReproductionCallback(ScanResult result) { }
        private void errorReproductionCallback(BluetoothDeviceStatus status) { }
        private static byte[] StringToByteArray(string hex)
        {
            return Enumerable.Range(0, hex.Length)
                             .Where(x => x % 2 == 0)
                             .Select(x => Convert.ToByte(hex.Substring(x, 2), 16))
                             .ToArray();
        }
    }
```

```

       public async Task connect(Guid deviceGuid, callbackBluetoothDeviceStatus callbackstatus)
        {
            try
            {
                var device = adapter.GetKnownDevice(deviceGuid);

                if(callbackstatus != null)
                {
                    device.WhenStatusChanged().Subscribe(status =>
                    {
                        if(status == ConnectionStatus.Connected)
                        {
                            callbackstatus(BluetoothDeviceStatus.connected);
                        } else 
                        if(status == ConnectionStatus.Disconnected)
                        {
                            callbackstatus(BluetoothDeviceStatus.disconnected);
                        }

                    });
                }
                await device.Connect();
            } catch (Exception e)
            {
                throw e;
            }
public BluetoothOperationResult disconnectDevice(Guid deviceGuid)
        {
            try
            {
                var device = adapter.GetKnownDevice(deviceGuid);
                device.CancelConnection();
                return BluetoothOperationResult.success;
            }
            catch (Exception e)
            {
                throw e;
            }
        }

public async Task writeCharac(Guid deviceGuid, Guid serviceGuid, Guid characertisticGuid, byte[] msg)
        {
            try
            {
                var device = CrossBleAdapter.Current.GetKnownDevice(deviceGuid);
                var service = await device.GetKnownService(serviceGuid);
                var characteristic = await service.GetKnownCharacteristics(characertisticGuid);
                var result = await characteristic.Write(msg);
            }
            catch (Exception e)
            {
                throw e;
            }
        }

        public async Task<Byte[]> readCharac(Guid deviceGuid, Guid serviceGuid, Guid characertisticGuid)
        {   
            try
            {
                var device = adapter.GetKnownDevice(deviceGuid);
                var characteristic = await device.GetKnownCharacteristics(serviceGuid, characertisticGuid);
                var result = await characteristic.Read();
                return result.Data;
            }
            catch (Exception e)
            {
                throw e;
            }
        }
```

aritchie, 2018-03-28:
Unfortunately, you've provided a sample that I can't just slap into a unit test and change some UUIDs.  When providing a reproducible sample, you need to strip away your abstraction code and show the raw issue.

Also, I've requested the FULL exception so I can track down the order of operations causing the issue.

Nebudef, 2018-03-28:
I would like to give you more as an Error message, but that just all i get.
Here is full running Code without any abstractions who produces the Error

```
private static byte[] StringToByteArray(string hex)
        {
            return Enumerable.Range(0, hex.Length)
                             .Where(x => x % 2 == 0)
                             .Select(x => Convert.ToByte(hex.Substring(x, 2), 16))
                             .ToArray();
        }
        private void rep()
        {
            Guid guid = new Guid("00000000-0000-0000-0000-eb806e44cfbc");
            Guid JAALEE_BEACON_SERVICE = Guid.Parse("0000fff0-0000-1000-8000-00805f9b34fb");
            Guid BEACON_KEEP_CONNECT_CHAR = Guid.Parse("0000fff1-0000-1000-8000-00805f9b34fb");

            var task2 = new Task(() =>
            {
                TimeSpan span = TimeSpan.FromSeconds(5);
                CrossBleAdapter.Current.ScanInterval(span).Subscribe(result => { });
            });
            var task = new Task(async () =>
            {
                while (true)
                {
                    try
                    {
                        var connectdev = CrossBleAdapter.Current.GetKnownDevice(guid);
                        await connectdev.Connect();

                        var device = CrossBleAdapter.Current.GetKnownDevice(guid);
                        var service = await device.GetKnownService(JAALEE_BEACON_SERVICE);
                        var characteristic = await service.GetKnownCharacteristics(BEACON_KEEP_CONNECT_CHAR);
                        var result = await characteristic.Write(StringToByteArray("666666"));

                        var device2 = CrossBleAdapter.Current.GetKnownDevice(guid);
                        var characteristic2 = await device.GetKnownCharacteristics(JAALEE_BEACON_SERVICE, BEACON_KEEP_CONNECT_CHAR);
                        var result2 = await characteristic.Read();

                        var device3 = CrossBleAdapter.Current.GetKnownDevice(guid);
                        device3.CancelConnection();
                    } catch(Exception e)
                    {
                        throw e;
                    }
                }
            });
            task.Start();
            task2.Start();
        }
```

aritchie, 2018-03-28:
You are abusing what GetKnownDevice is used for.  The device registry will clear with each scan.  

```csharp
// keep the device reference below!
var connectdev = CrossBleAdapter.Current.GetKnownDevice(guid);

// if you are having issues between scans, this may return null
if (connectdev != null)
{
    await connectdev.Connect();

// don't keep calling this, store it locally, make sure to clear between connections - getknowncharacteristics returns a list - you should throw timeouts on this just in case it can't be found
    var characteristic = await device.GetKnownCharacteristics(JAALEE_BEACON_SERVICE, BEACON_KEEP_CONNECT_CHAR).Take(1).ToTask();    

   var read = await charateristic.Read();
   device.CancelConnection();
}
```

Nebudef, 2018-03-29:
that helped, thank you

closed 2018-03-29 by Nebudef