aritchie/bluetoothle #161: Existing Scan

Uniphix, 2018-03-29:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [x] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
11.2.16 iOS

## Version of Library
5.3.4

## Expected Behaviour
I have basically used the Scan() method.  However if I use a Scan() when the Adapter has an existing scan (due to a crash) or a (restart) of the application.  It will throw "Existing Scan" already exists or something like that.  How can I rebind that scan?

aritchie, 2018-03-29:
I've created a StopScan method that only exists in beta at the moment.  I'll look into backporting that in v5.  

I'm working on ways to participate in the existing scans going forward.

cunnza, 2018-04-06:
hello,

same bug also exists in beta unfortunately. This time not only on iOS.

aritchie, 2018-04-06:
@bluekuen I'm only taking PR's on the betas at this point.  The issue in any case has nothing to do with this one