aritchie/bluetoothle #163: After re-connect WhenCharacteristicDiscovered() does not work on devices retrieved by CrossBleAdapter.Current.GetConnectedDevices()

mstein11, 2018-03-30:
## Please check all of the platforms you are having the issue on (if platform is not listed, it is not supported)

 - [X] iOS
 - [] Android
 - [] tvOS
 - [] macOS
 - [] .NET Standard
 - UWP IS NOT SUPPORTED DURING THE BETA PERIOD - ONLY PRs ARE WELCOME FOR NOW

## Version of OS(s) listed above with issue
11.2.6

## Version of Library
5.3.5

## Expected Behaviour
1. Scan for BT Device
2. Connect to BT Device
3. Write to BT Device
4. BT Device becomes out of range
5. WhenStatusChanged() indicates loss of connection (disconnected)
6. Devices is in range again
7. CrossBleAdapter.Current.GetConnectedDevices() should not return the specific device.
8.  Scan for BT Device
9. Connect to BT Device
10. Write to BT Device

## Actual Behavior
1. Scan for BT Device
2. Connect to BT Device
3. Write to BT Device
4. BT Device becomes out of range
5. WhenStatusChanged() indicates loss of connection (disconnected)
6. Devices is in range again
7. CrossBleAdapter.Current.GetConnectedDevices() Returns device that is now in range again (WhenStatusChanged was not triggered, indicating it is still disconnected)
8. on that device WhenCharacteristicDiscovered() does not find any characteristics. (Therefore we can not write)

## Code of Conduct
* No Library Update Requests
* No Timeline Requests
* Please read through the documentation and FAQs
* Please be respectful of my time. I put a lot of effort into my libraries.  I try to support the issue requests within a day.  I'm not psychic, so please take the time to fill out the above form as complete as possible

aritchie, 2018-03-30:
Please submit a small reproducible sample. 

Also, if you are trying to manage rescanning and reconnection yourself, the simple answer is - don't.  Keep a reference to the device and wait for WhenStatusChanged to refire.  I'm not sure if that's what you're stating your doing in point 7 or not.

mstein11, 2018-04-02:
I will try to provide a reproducible sample, however it might take some time until I can do so. 

Regarding your suggestion, not to handle the rescanning and reconnecting myself, I tried to let the library handle those things. But for this particular issue, it seems that the reconnect event is not properly fired (as explained in the above). As a workaround, I manually rescan for the particular device (only when CrossBleAdapter.Current.GetConnectedDevices() returns a device that is not able to discover characteristics). Do you see any problem with doing so? What is the issue with handling the reconnection myself in this case?

What also might be interesting is that when the issue occurs CrossBleAdapter.Current.GetConnectedDevices() returns a device that has the status "connecting" instead of connected. Also the device seems to be stuck at the "connecting" status, it doesn't change from there to "connected" or "disconnected". 

Anyway, thanks for the quick response and for maintaining this project!

aritchie, 2018-04-14:
Fixed in v6 beta4

closed 2018-04-14 by aritchie