aritchie/bluetoothle #164: Notifications are not fired ONLY on Android 7.x and 8.x

kikutano, 2018-03-31:
I'm using your Plugin and everything works great, no problem to read the measurements and no issue until I try to use my application on **Android 7.0**, **7.1** and **8.0**, **8.1**. For some reason, the same code that's works on **Android 6.0.1** doesn't works on **Android 8.1**. On Android 8.1 I can connect without problem, the services are discovered well, **_but the Notifications are not fired_** and I can't understand why. Here my code: [https://pastebin.com/qCa2P4uZ](https://pastebin.com/qCa2P4uZ ), on Android 8.1 everything works great until `cD.RegisterAndNotify().Subscribe( registerResult => {` at line 81, the line 83 is never called, but only on Android 7/8, on Android 5/6 no problems. It's seems that the Notifications can be enabled or fired on the new Android OS 8.1.

Any suggests? Where I'm wrong in my code? 

Thanks.

aritchie, 2018-03-31:
Use the issue template in the future and do not open identical issues to what you have opened in the past please.  

1) get rid of the RSSI check on the device.  It often interrupts android connections.
2) Dispose of the scanner before creating the connection
3) I wouldn't do this - if ( sD.Description == "Cycling Speed and Cadence" ) {    Figure out your service UUID and use something like device.GetKnownService(UUID)
4) Or futher to 3, get the specific characteristic you need as well - device.GetKnownCharacteristic(serviceUUID, characteristicUUID).  
5) Lastly, in your android project, try setting CrossBleAdapter.AndroidPerformActionsOnMainThread = true  (you won't see this flag in your netstandard project, only on your android project).  This only sets to true for older devices or samsung devices.  It will be on by default in the next release.

All of the techniques above are designed to take pressure off the android radio which is easy to overwhelm if you read/write something before a previous operation has finished.

kikutano, 2018-03-31:
Thanks you for your reply, I tried your suggests but the result it's the same. Did you ever try on Android >7.0? Nobody had report this kind of issue on Android >7.0? I will try to continue to test to understand where the problem is.

aritchie, 2018-03-31:
Yes - I have tested this on several devices and OS versions.  You are welcome to scan the issues in github to see this!  The bugs are in the BLE android stack and the fact that it isn't thread safe.  I have investigated ways to fix this but only ended up causing different issues.  With that, I have chosen to open up the android API for others to fight with.  Usually, if you follow the suggestions above, everything will go fine.  I also run this library in a few projects without issue.

kikutano, 2018-04-03:
I compiled you example on Android 8.1 and it works great. So, the problem is in my code, I'm trying to understand where and I will report here my progress as soon as I can.

kikutano, 2018-04-03:
I fixed the problem on Android 8. For some reason, only on Android 7.x/8.x a characteristic that I don't need has `.CanNotify()` as true and this is not a notification Characteristic, while on Android <6 the same Characteristic had `.CanNotify()` as false. So my code fire `RegisterAndNotify()` where is not need and everything goes wrong. Now I'm getting the Characteristics with the Id and everything is working on Android 8! You can close this issue! Thanks!

closed 2018-04-08 by aritchie