aritchie/bluetoothle #167: could you maybe consider using a lower but sufficient netstandard version instead of 2.0

BitBender1, 2018-04-15:
first let me thank you for the terrific work, i cant imagine how much time you must have spent in making this library.

now to our issue:
some versions ago you switched to .netstandard 2.0 and this forces us to either lose compatibility with lots of devices, or we stick with an older version of the library, not getting access to all improvements and bugfixes that came afterwards.

to my understanding the idea behind .netstandard would be that every version is just an agreement on a set of APIs, where higher versions include a bigger API surface. it is not like with software-versions where higher versions are by design better or more advanced or more performant, its really just an API list that gets longer with each version. so idea would be that a library uses the lowest possible .netstandard-version that supports all APIs and features really used and required by that library.

as far as i can see your bluetooth-library should work fine without restrictions with .netstandard 1.1 or 1.2, making it a lot more backwards-compatible without any need to sacrifice anything. i certainly do not know if you plan to use APIs in the near future that only exist in .netstandard 2.0, but if you dont it would make sense and help us - and probably also others - if you could reduce the .netstandard-version to whatever it really needs to be.

issue we face is that on windows .netstandard 2.0 only exists from fall-creators-update upwards, meaning we immediately lose access to about at least two thirds of devices if we switch to .netstandard 2.0. 

your library works impressively well with UWP, but it will take quite a while until all computers will be on that level to support .netstandard 2.0.

so as your library makes no use of the APIs and there is no obvious reason to require .netstandard 2.0 i would ask you to consider adjusting the version to what you really need, and not to whatever version is currently theoretically available.

thanks for your time :)

aritchie, 2018-04-15:
Sure - I can definitely bring back .net standard 1 in v6 of the library where I have multitargeting.

BitBender1, 2018-04-15:
great! thx

BitBender1, 2018-04-16:
i certainly do not know what future plans you have for the library, but i must admit i do not understand why multitargeting would be a requirement to pick a lower netstandard version. multitargeting for nuget-libraries was invented to enable support for different and not really matching framework-implementations like to support full .net-framework and UWP. ( [https://docs.microsoft.com/en-us/nuget/create-packages/supporting-multiple-target-frameworks](https://docs.microsoft.com/en-us/nuget/create-packages/supporting-multiple-target-frameworks) )

in contrast netstandard is not an implementation at all, but just an agreement on supported APIs, where by definition a lower netstandard-version is a well defined and fixed subset of each higher version. by picking a version, the compiler as well as visual studios intellisense takes care that a developer does not escape the self-imposed version-restriction by using an API thats not part of the contract. but picking a higher version than is required to have access to the APIs used in the code only reduces the number of machines it will run on.

so netstandard 2.0 is no different target to netstandard 1.1 in any way, by targeting 1.1 or 1.0 or 1.2, a library will by design also work perfectly as intended on any machine supporting netstandard 2.0 and above.

i would like to reference this microsoft-documentation-page: [https://docs.microsoft.com/en-us/dotnet/standard/net-standard
](https://docs.microsoft.com/en-us/dotnet/standard/net-standard)

regarding versioning and how to pick an adequate netstandard-version, the microsoft documentation referenced above says the following:
"In general, we recommend you to target the lowest version of .NET Standard possible. So, after you find the highest .NET Standard version you can target, follow these steps:
Target the next lower version of .NET Standard and build your project.
If your project builds successfully, repeat step 1. Otherwise, retarget to the next higher version and that's the version you should use."

maybe you have other reasons or future plans to really need netstandard 2.0, but if you dont it is really only preventing that great and useful library from running on perfectly capable platforms. netstandard is not like framework-versions where using the latest is a meaningful goal. the version of netstandard really only is meant to declare what set of APIs your code needs to run, and code that uses APIs that are completely covered in the definition of netstandard 1.1 should even in 15 years declare its needs with netstandard 1.1. by needlessly picking 2.0 you currently only demand that a platform supports and implements a ton of APIs that the code in reality does not use or need.

aritchie, 2018-04-16:
Trust me, I understand all of the points you are making since I've been doing the cross platform thing for a while now.  I have reasons for wanting to push netstandard 2.  Some are political, some are defects that popup through xamarin.

BitBender1, 2018-04-16:
makes sense, thanks for the explanation and your patience!

aritchie, 2018-04-17:
While I appreciate that you would like to have netstandard 1.  I have to add an annoying little package called Microsoft.NETCore.UniversalWindowsPlatform to get anything to compile properly (netstandard isn't the problem - UWP is which is what you're looking).  The problem is that the difference in target versions also changes this library.  I'm looking into multitargeting with UWP versions as well, but it is likely going to come to the fact that it simply isn't worth it to me.  The fact is, UWP will likely never leave beta in this library due to the way MSFT implemented their crappy API and I've contacted them via several channels without any sort of response. 

Sorry.

closed 2018-04-17 by aritchie